victronenergy/gui-v2 #1828: Settings -> ESS -> Grid feed-in goes to -1, 199, 299 etc Watts

mpvader, 2025-01-11:
While it should be 0, 100, 200, …

I was able to reproduce this on Venus OS v3.60~15 within a few minutes, on this site:

https://vrm.victronenergy.com/installation/473156/firmware-update

to reproduce it, press the down button till you see -1.
 
[image]

reported here:
> I noticed the grid feed in limit goes to -1 instead of 0 then when you increase the limit again it goes in steps of 99 watts instead of 100 watts.

https://community.victronenergy.com/t/venus-os-v3-60-15-available-for-public-testing/13471

MikeTrahearn-Qinetic, 2025-01-13:
In this case the specific ListSpinBox "from" property is not explicitly set to 0; rather it follows the VEQuickItem's min property, which in this case is reporting -1 from the installation.

MikeTrahearn-Qinetic, 2025-01-15:
@DanielMcInnes noted that gui-v1 specifies the min value as 0 so perhaps we should do the same here then.

commit 46e3cde0 (2025-01-15): Fix Maximum ESS Grid Feed-in could be set to -1

As per Gui-v1, the maxFeedInPower ListSpinBox 'from' property
should be set to 0.

Fixes #1828
PR #1846 (2025-01-16): Fix Maximum ESS Grid Feed-in could be set to -1
As per Gui-v1, the maxFeedInPower ListSpinBox 'from' property should be set to 0.

Fixes #1828
commit dd070639 (2025-01-16): Fix Maximum ESS Grid Feed-in could be set to -1

As per Gui-v1, the maxFeedInPower ListSpinBox 'from' property
should be set to 0.

Fixes #1828
closed 2025-01-16 by MikeTrahearn-Qinetic