victronenergy/gui-v2 #1830: Tank settings and Levels page shows incorrect tank capacity when greater than 9999 litres

blammit, 2025-01-12:
As per screenshots:

[image]
[image]

blammit, 2025-01-13:
Note: also check the font rendering, as the "l" litres symbol is not rendering as expected (screenshot was from Safari).

chriadam, 2025-02-10:
I suspect it's trying to use kL as the units rather than L, but perhaps something is going wrong with the unit display?

chriadam, 2025-02-10:
I tested this (with mock mode) on both desktop (windows) and cerbo gx (arm).  In both cases, the unit renders correctly as kL, see:

[image]

chriadam, 2025-02-10:
But, with wasm, we see that it does NOT render properly:

[image]

So, this sounds like a font issue to me - the glyph for "kl" doesn't exist in the font used in the web assembly, perhaps?
The unicode codepoint is QStringLiteral("\u3398").

chriadam, 2025-02-18:
PR#1900 was merged.  Closing this issue as fixed.

closed 2025-02-18 by chriadam