victronenergy/gui-v2 #1836: Redundant decimal places shown in tank sensor settings

blammit, 2025-01-13:
See screenshots below, as per slack thread: https://victrondevelopment.slack.com/archives/C04RYU0UMDE/p1736602047609179

In the last production release of gui-v2, the tank settings page looks like this, with only 1 decimal places for `cm`:

[image]

But in the latest beta there are redundant zeros:

[image]

DanielMcInnes, 2025-01-16:
Screenshots from installation 60999

mpvader, 2025-01-18:
Hey Daniel, can you make sure that the sensor value field uses the same number of decimals?

So for cm its 1 and for all others its three.

mpvader, 2025-01-23:
hey @DanielMcInnes looks like you missed my comment that I made in the issue, can you make this change as well?

DanielMcInnes, 2025-01-24:
@mpvader will do, there's a review up now

commit ea3b0942 (2025-01-16): Change the number of decimals used to display tank sensor values when empty / full

Fixes #1836
commit 9240c8b9 (2025-01-17): Change the number of decimals used to display tank sensor values when empty / full

Fixes #1836
commit 9aeed323 (2025-01-20): Change the number of decimals used to display tank sensor values when empty / full

Fixes #1836
commit 50951a01 (2025-01-20): Change the number of decimals used to display tank sensor values when empty / full

Fixes #1836
closed 2025-01-24 by DanielMcInnes