victronenergy/gui-v2 #1850: The arrow in SettingsListNavigation is not vertically centred

DanielMcInnes, 2025-01-16:
[image]

MikeTrahearn-Qinetic, 2025-03-14:
Aside observation - SettingsListNavigation shadows the secondaryText property alias from ListNavigation.

MikeTrahearn-Qinetic, 2025-03-14:
This was quite tricky to get to the bottom of because there were three layers of layouts competing with each other for supremacy. After analysis of SettingsListNavigation, ListNavigation and ultimately ListItem, it was discovered that the GridLayout inside the ListItem providing the container for the Row in which the ListNavigation's arrow icon would sit, was not responding according to the change of **explicit** height being applied to the ListItem via SettingsListNavigation. A fix that would not harm any normal implicit-height ListItems but would fix SettingsListNavigation was to make the GridLayout (which fills the parent) also vertically center in it.

MikeTrahearn-Qinetic, 2025-03-14:
Turns out that this simple fix isn't correct after all because the layers of complexity between ListRadioButtonGroup, through ListNavigation to ListItem where there are multiple overlapping rules about the layout and sizing policies.

MikeTrahearn-Qinetic, 2025-03-14:
For the first issue: Fixed the ListItem sizing to ensure it follows the implicitHeight of the GridLayout (contentLayout) in the case that the ListItem interactive is false, making the icon vanish and cause the ListItem implicitHeight to be miscalculated.

For the latter issue (overlapping): the case in PageDebug.qml (and also one other discovered case in TemperatureRelaySettings.qml) had an additional Label simply stuck on top of the ListRadioButtonGroup. Both of these cases have been replaced by utilising the ListItem's "caption" property which places the additional text in the bottomContent, and the ListItem resizes correctly to accomodate the extra caption text when provided.

commit 889df1cb (2025-03-14): Fix vertical centering of arrow icon in SettingsListNavigation

- Fix ListItem's GridLayout (contentLayout) to centerIn: parent
so that any changes in the explicit height of ListItem (as applied by
SettingsListNavigation) will result in the contentLayout remaining
vertically centered in the entire ListItem.
- Fix ListItem's height to use the GridLayout's implicitHeight

Fixes #1850
commit 3533c223 (2025-03-14): Fix vertical centering of arrow icon in SettingsListNavigation

- Fix ListItem's GridLayout (contentLayout) to centerIn: parent
so that any changes in the explicit height of ListItem (as applied by
SettingsListNavigation) will result in the contentLayout remaining
vertically centered in the entire ListItem.
- Fix ListItem's height to use the GridLayout's implicitHeight
- Fix overlapping text in ListRadioButtonGroup

Fixes #1850
PR #1979 (2025-03-18): Fix vertical centering of arrow icon in SettingsListNavigation
- Fix ListItem's GridLayout (contentLayout) to centerIn: parent so that any changes in the explicit height of ListItem (as applied by SettingsListNavigation) will result in the contentLayout remaining vertically centered in the entire ListItem.
- See ticket for explanation of how the problem was distilled into this tiny fix.

Fixes #1850
commit 51c1105d (2025-03-18): Fix vertical centering of arrow icon in SettingsListNavigation

- Fix ListItem's GridLayout (contentLayout) to centerIn: parent
so that any changes in the explicit height of ListItem (as applied by
SettingsListNavigation) will result in the contentLayout remaining
vertically centered in the entire ListItem.
- Fix ListItem's height to use the GridLayout's implicitHeight
- Fix overlapping text in ListRadioButtonGroup

Fixes #1850
closed 2025-03-18 by MikeTrahearn-Qinetic