victronenergy/gui-v2 #1854: Could add an icon or other indicator in animations-paused mode (high-cpu)

chriadam, 2025-01-16:
Otherwise the user might think that the paused / stopped animations are unintentional

chriadam, 2025-11-06:
Pushed back to v1.3.0 milestone.  @philipptrenz maybe we could add a "turtle" icon in the status bar when animations are paused due to high cpu load, if Serj could provide such an icon?  Anyway, low prio for now.

ReinvdZee, 2026-10-20:
A user on community again reported that after updating the animations stopped working: https://community.victronenergy.com/t/venus-os-v3-90-beta7-energy-flows-on-display-bug/64057

Increasing the prio on this one because its important to inform the user that stopping the animations is expected under high cpu load.

ReinvdZee, 2026-10-20:
Added require docs label to make sure this behavior is in the manual

ReinvdZee, 2026-10-22:
Design ready: https://www.figma.com/design/GjCtz3A3jqSGvyjVvOm6RZ/Venus-OS--GX-Touch-?node-id=22143-505482&t=yY5rB2f1e5asbADH-4

Serj has designed this 'snail' icon (very creative!) to indicate that animations are stopped:

[image]

Furthermore, there should be a toast notification when animations are disabled:

[image]

Personally I think the notification should say something like:
"Animations paused due to high CPU load. Values are still updating"

The toast should be triggered when animations are disabled, or when the user presses the snail icon. It should be dismissed after 5 seconds or if the user presses the 'X'.