victronenergy/gui-v2 #1855: Investigate issue re. rebooting cerbo causing watchdog timeouts

DanielMcInnes, 2025-01-16:


DanielMcInnes, 2025-01-16:
@mr-manuel has seen this happen

mr-manuel, 2025-01-23:
How to reproduce: Open the GUIv2 in the browser and update the Cerbo GX or reboot it. If it takes too long the page refreshes and says the server is not found. If you want to test it the hard way just disconnect the PC or GX device from the network and wait.

You could check, if the server is reachable before refreshing the page.

DanielMcInnes, 2025-01-24:
Cannot reproduce. I connected to my cerbo via vrm, then powered off my cerbo for a few minutes, and I didn't see the page refresh. @jepefe , what is your use-case for kiosk-mode android builds? Do you connect via vrm?

mr-manuel, 2025-01-24:
Sorry, this problem is not via VRM. It's if you open the local IP/hostname and GUIv2.

[image]

DanielMcInnes, 2025-01-27:
I can reproduce this behavior, but I don't think it is a real issue. The watchdog is working as intended. If I run a firmware update from a wasm build, I am not really surprised when I get a 'page not found' error. Once the cerbo reboots, I can refresh the web page, and it reloads ok.

mr-manuel, 2025-01-27:
If you have a display where you are displaying the GUIv2 as full screen in a browser this is an issue. I can open a PR with a fix. Before reloading the page, a simple check that checks, if a HTTP call was successful is enough and only then refresh the page, else wait.

mr-manuel, 2025-02-06:
For reference: https://github.com/victronenergy/gui-v2/pull/1801

commit f6617a15 (2025-01-27): Use already present reload function to reload page
Fixes #1855
PR #1906 (2025-02-17): Use already present reload function to reload page
Fixes #1855
commit 8ad0a6c9 (2025-02-17): Use already present reload function to reload page
Fixes #1855
closed 2025-02-17 by chriadam