victronenergy/gui-v2 #1856: Show QR/Link code to add GX to VRM

jepefe, 2025-01-16:
This allows to register the GX in VRM without having to connect with VC via Wifi/Bluetooth. 

This url will open the VRM add installation page if the GX is not registered yet, otherwise it will open the installation page.

"https://vrm.victronenergy.com/login?installationIdentifier=" + vrmId + "&device=" + deviceType

```
deviceType = "ccgx" // Color Control GX
deviceType = "vgx" // Venus GX
deviceType = "ogx" // Octo GX
deviceType = "egx" // EasySolar-II GX
deviceType = "mgx" // MultiPlus-II GX
deviceType = "ekranogx" // Ekrano GX
deviceType = "cerbogx" // Cerbo GX
```

blammit, 2025-01-16:
@jepefe where should this link be placed, and what should be the title/name of the link?

blammit, 2025-01-19:
This can go on the "VRM" page (between VRM Portal ID and Log interval), with "Register installation" as the title text.

mpvader, 2025-01-21:
I've updated the first message to the outcome of the discussion wrt how to tackle this + marked it as blocked since its awaiting some changes elsewhere.

mr-manuel, 2025-01-21:
With my current modifications `com.victronenergy.platform/Device/Model` is added, if that is of relevance.

Output:
- `Cerbo GX`
- `Ekrano GX`
- `MultiPlus-II`

https://github.com/victronenergy/venus-platform/blob/043d7a8768559ed954cba9c69ecd8808c9efae5e/src/application.cpp#L681-L684

ReinvdZee, 2025-01-21:
Changes pushed and on the todo.
The product ID will be published on `com.victronenergy.platform/Device/ProductId`.

mpvader, 2025-01-30:
This is now waiting for the vrm api to have been updated

AlexImb, 2025-01-30:
> Exact URL to be confirmed by Alex

URL structure will be: https://vrm.victronenergy.com/login?installationIdentifier=" + vrmId + "&productId=" + deviceType

Implementation will be live next week.

chriadam, 2025-05-16:
Merged to main, closing as fixed.

closed 2025-05-16 by chriadam