victronenergy/gui-v2 #1865: VE.Bus: Inverter control card issues

vejpasop, 2025-01-20:
On a VE.Bus inverter only model such as a "Phoenix Inverter Compact 12/1200| GUI V2 shows "Inverter/ **Charger**" in the control card. Also the Charger only and Inverter only options are still shown. (I notice that the selections are grayed out so the logic to detect if the VE.Bus device is a inverter probably already is in place).

The "Charger" text and "Charger only"/"Inverter only" options should not be visible when connected to an inverter only model. See GUI V1

[image]

[image]

For completeness: A VE.Bus inverter can be spotted when under `com.victronenergy.vebus.tty##` the item value of `Ac/NumberOfAcInputs ` equals 0,

Here is a dbus recording of a inverter

[vebus_inverter.csv](https://github.com/user-attachments/files/18478185/vebus_inverter.csv)

blammit, 2025-01-20:
@vejpasop looks like this is not a critical issue and shouldn't be in v3.50-fixes?

vejpasop, 2025-01-21:
Hi @blammit. 
I added it to the v3.50-fixes by request of @mpvader

blammit, 2025-01-21:
@vejpasop Ah I see, okay thanks 👍

commit 48bd44f7 (2025-01-22): Don't display 'Inverter/Charger' for inverter-only devices

Fixes #1865
commit 79c068d5 (2025-01-22): Don't display 'Inverter/Charger' for inverter-only devices

Fixes #1865
commit 5ec83e77 (2025-01-22): Don't display 'Inverter/Charger' for inverter-only devices

Fixes #1865
commit e5175841 (2025-01-22): Don't display 'Inverter/Charger' for inverter-only devices

Fixes #1865
closed 2025-01-24 by DanielMcInnes