victronenergy/gui-v2 #1870: Solar Power Rendering issue on Raymarine Axiom 12

mman, 2025-01-21:
While working on https://github.com/victronenergy/venus-html5-app/issues/467 I have left GUI v2 running some time and it currently displays ghost solar power stripe. The other three stripes are rendered correctly so there must be some different arithmetics happenings rendering different layers of the bottom left stripe...

Opening the page in another browsers renders it just fine...

Reloading the GUI on Raymarine also shows it rendering correctly at the beginning, but over time it breaks...

[image]

chriadam, 2025-03-03:
Do you mean the fact that the lower-left gauge has 3 arcs?  I suspect it's intentional, because we show historical data there.  See SolarYieldGauge.qml for more information.  Briefly, a comment there reads:
`Take 30-second samples of the solar power. Every 5 minutes, take the average of these samples and add a new gauge bar with that value.`

mman, 2025-03-20:
See also report from the community here: https://community.victronenergy.com/t/v3-60-46-brief-page-shows-3-bars-for-solar-on-system-with-one-mppt/25911

blammit, 2025-03-20:
Indeed those extra bars are intentional; they show the recent solar history on the system, but yes it's not clear to the user what those bars are supposed to represent. Perhaps that could be improved in a later release.

FYI @mpvader @philipptrenz

mman, 2025-03-21:
Hmmm... that is interesting. I would swear when this happened on my device, I connected from a browser and that did not show the three stripes... so I immediately jumped to conclusion that it's a rendering problem... looks like I must have missed that.

In any case, it definitely is not clear what they should represent.

chriadam, 2025-11-06:
Updated bug title, and pushed back to v1.3.0, @philipptrenz this one needs discussion in future with you and Serj I think, but low prio for now.