victronenergy/gui-v2 #1881: gui-v2 WASM crashes when changing the security profile from Unsecured to Weak

mpvader, 2025-01-29:
And maybe when going from other profiles to other profiles as well. The setting does get changed effectively, so for the user its a simple matter of restarting the gui-v2 as a wasm. (clicking the link on VRM). He/she is not stuck.

Found while triaging a report here: https://victrondevelopment.slack.com/archives/C04CDB8KT0T/p1738139369132709.

This should be super simple to reproduce, I reproduced it in 5 seconds.

If easy to fix and backport, then lets do that. But if complicated, then OK for me to leave it as is and fix only in master cq. Venus OS v3.60.

blammit, 2025-01-30:
The Wasm app does have to reload when the security profile is changed, but it should not crash. I tested this with the latest `main`: if I connect with Wasm directly via MQTT on the local server, then the app reloads as expected after the profile change; if I connect via VRM portal, then the app quits after the profile change.

mpvader, 2025-01-30:
Ok, then how about adding a notification that it will reload, as well as fixing the reload on vrm?

I’ll change the prio to medium

closed 2025-02-25 by DanielMcInnes