victronenergy/gui-v2 #1882: Loading HTML screen not aligned properly

mman, 2025-01-29:
When the WASM initial loading screen with Victron logo is painted on screen... it's not aligned properly vertically...  and remains near the top of the screen.... where after several seconds (when 33MB gzipped wasm is loaded) it is replaced with centered Victron logo and MQTT loading progress bar... that are however centered nicely... to fill available space...

This can be seen on initial connection to GUI-v2 via browser that has not yet downloaded the WASM, and it is very visible on slower devices like Raymarine Axiom 12 where loading takes several seconds...

nice to have to make it look better...

[image]

[image]

mpvader, 2025-06-05:
discussed today: Manuel is working on something that fixes this. He'll make sure its also tested on an MFD and then close this

mr-manuel, 2025-06-16:
Will be fixed with https://github.com/victronenergy/gui-v2/pull/2236

[image]

mman, 2025-06-16:
That looks much better @mr-manuel, thanks for addressing this!

mr-manuel, 2025-06-16:
You're welcome

PR #2236 (2025-07-25): Add loading bar for WASM
Fixes #2299

Fixes #2229

Fixes #1882