victronenergy/gui-v2 #1883: No Voltage and Current displayed in Overview > Solar Yield

SenH, 2025-01-30:
### `Overview > Solar Yield`
[image]

### Voltage/Current is displayed in `Settings > Devices > Solar`
[image]

I updated today to Venus v3.54 but this has never been displayed since GUIv2 was introduced.

blammit, 2025-01-31:
This can be tested on installations with a single-phase PV inverter.

May be related to https://github.com/victronenergy/gui-v2/issues/1713.

chriadam, 2025-02-05:
According to https://github.com/victronenergy/venus/wiki/dbus#pv-inverters the `/Ac/Current` and `/Ac/Voltage` paths are deprecated, and presumably systemcalc isn't publishing appropriate values there, but we use those in PvInverter.qml.  For single-phase inverters, we can just find and use the appropriate per-phase data e.g. `/Ac/L1/Current` and `/Ac/L1/Voltage` directly, however for multi-phase systems I am not sure what we can do (voltage cannot be summed, and summing separate phases of AC current makes no sense either).  Well, I will fix the single-phase case now, and we can discuss the multiphase case further in issue 1713.

chriadam, 2025-02-18:
Merged PR#1892 to resolve this issue.  Closing as fixed.

closed 2025-02-18 by chriadam