victronenergy/gui-v2 #1896: Error running WASM builds via vrm 'remote console' with weak security profile

DanielMcInnes, 2025-02-05:
If I launch a wasm build via vrm 'remote console' (i.e. by clicking on the link in vrm, not by running a localhost build) while the security profile set to 'Unsecured' or 'Secured', it loads fine. If I then change the security profile to 'Weak', I get an error loading the WebAssembly.

console log:
```MQTT client error: QMqttClient::TransportInvalid app.c6f584c2.v1.js:9:58148
BackendConnection state: Victron::VenusOS::BackendConnection::Disconnected app.c6f584c2.v1.js:9:58148
qml: Rebuilding UI app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/SystemLoad.qml:32: TypeError: Cannot read property 'input1Info' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/EvChargers.qml:35: TypeError: Cannot call method 'chargerModeToText' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/common/Generator.qml:19: TypeError: Cannot call method 'runningByText' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/common/Generator.qml:20: TypeError: Cannot call method 'stateText' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/common/Generator.qml:34: TypeError: Cannot call method 'isAutoStarted' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/common/Notification.qml:13: TypeError: Cannot read property 'serviceUid' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/System.qml:58: TypeError: Cannot read property 'model' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/SystemLoad.qml:47: TypeError: Cannot read property 'veBus' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/SystemLoad.qml:50: TypeError: Cannot read property 'veBus' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/common/PvMonitor.qml:51: TypeError: Cannot read property 'serviceUid' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/common/AcInputSystemInfo.qml:16: TypeError: Cannot read property 'serviceUid' of null 2 app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/AcInputs.qml:55: TypeError: Cannot read property 'serviceUid' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/AcInputs.qml:59: TypeError: Cannot read property 'serviceUid' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/System.qml:46: TypeError: Cannot read property 'serviceUid' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/System.qml:79: TypeError: Cannot read property 'serviceUid' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/System.qml:113: TypeError: Cannot read property 'serviceUid' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/System.qml:117: TypeError: Cannot read property 'serviceUid' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/DcInputs.qml:122: TypeError: Cannot read property 'serviceUid' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/common/AcInputSystemInfo.qml:54: TypeError: Cannot read property 'serviceUid' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/common/AcInputSystemInfo.qml:58: TypeError: Cannot read property 'serviceUid' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/common/AcInputSystemInfo.qml:54: TypeError: Cannot read property 'serviceUid' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/common/AcInputSystemInfo.qml:58: TypeError: Cannot read property 'serviceUid' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/common/EssData.qml:85: TypeError: Cannot read property 'serviceUid' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/common/EssData.qml:89: TypeError: Cannot read property 'serviceUid' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/common/EssData.qml:93: TypeError: Cannot read property 'serviceUid' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/common/EssData.qml:97: TypeError: Cannot read property 'serviceUid' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/VenusPlatform.qml:19: TypeError: Cannot read property 'serviceUid' of null app.c6f584c2.v1.js:9:58148
qrc:/qt/qml/Victron/VenusOS/data/SystemSettings.qml:178: TypeError: Cannot read property 'serviceUid' of null app.c6f584c2.v1.js:9:58148
qml: Rebuilding complete app.c6f584c2.v1.js:9:58148
QIODevice::write (WebSocketDevice): device not open app.c6f584c2.v1.js:9:58148
BackendConnection state: Victron::VenusOS::BackendConnection::Reconnecting app.c6f584c2.v1.js:9:58148
Removing unpermitted intrinsics lockdown-install.js:1:52832
[shared-components] Registering vue component vrmAddInstallation app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmAddInstallationDrop app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmAvatar app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmAvatarStack app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmBetaBadge app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmButtonDropdown app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmCircularProgress app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmCurrencyDropdown app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmAddInstallationHeader app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmDrawer app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmDropdownGx app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmEvents app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmGlobalMessage app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmInfoTooltip app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmInstallationAvatar app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmInstallationAvatarStack app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmInstallationCard app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmInstallationsOverviewFilterDropdown app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmInstallationsFilterIndicator app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmInstallationsTableRenderer app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmInstallationsTableRendererMobile app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmLocationPin app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmMapGeofenceSettings app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmMapGpsWidget app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmMapLocationSettings app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmModalPresenter app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmNavInstallationItem app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmNewBadge app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmNewFeaturesBanner app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmNotifications app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmOverviewMap app.c6f584c2.v1.js:9:633533
[shared-components] Registering vue component vrmPaginationList app.c6f584c2.v1.js:9:633533
[... 135 lines cut ...]
chunk-vendors.aaec8f4d.v1.js:407:2787
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at https://fcm.googleapis.com/fcm/connect/subscribe. (Reason: CORS header ‘Access-Control-Allow-Origin’ missing). Status code: 404.
[localizationService] Loading locale module 'en/alarms' chunk-vendors.aaec8f4d.v1.js:103:84157
[MqttService] webmqtt.victronenergy.com :: Disconnecting chunk-vendors.aaec8f4d.v1.js:103:84157
[MqttService] webmqtt58.victronenergy.com :: Connecting: webmqtt58.victronenergy.com chunk-vendors.aaec8f4d.v1.js:103:84157
[MqttService] webmqtt58.victronenergy.com :: Sending realtime keepalive for 48e7da88d711 chunk-vendors.aaec8f4d.v1.js:103:84157
[localizationService] Loading locale module 'en/general' chunk-vendors.aaec8f4d.v1.js:103:84157
[... cut ...]
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at https://150502-gui.proxyrelay3.victronenergy.com/gui-beta/venus-gui-v2.wasm. (Reason: CORS request did not succeed). Status code: (null).
Error: Could not fetch WebAssembly app.c6f584c2.v1.js:9:52109
Error: Could not fetch WebAssembly app.c6f584c2.v1.js:9:520385
XHRGET
https://updates.victronenergy.com/venus-gui/45979f571dcc327e8b6aff668bf12dc84bbde7d3fac4e1ba046ea9a4255eba11/venus-gui-v2.wasm
[HTTP/1.1 404 Not Found 30ms]

RC RunningVersion: 1 RC_GUI_V2 app.c6f584c2.v1.js:9:318221
```

chriadam, 2025-02-06:
Indeed.  I see the same thing.

Additional interesting lines from my console log follow:

```
Failed to load resource: the server responded with a status of 404 (Not Found) https://updates.victronenergy.com/venus-gui/b917ad9bdf3a7985687087cd61d59bbcfd592c144b82eafb25dcaf22445d0cca/venus-gui-v2.wasm
RC RunningVersion: 1 RC_GUI_V2
Mixed Content: The page at 'https://vrm.victronenergy.com/installation/133622/dashboard' was loaded over HTTPS, but requested an insecure resource 'http://133622-gui.proxyrelay3.victronenergy.com/auth/login.php?page=/gui-beta/venus-gui-v2.wasm'. This request has been blocked; the content must be served over HTTPS.
qtloader.js:258 Error: Could not fetch WebAssembly
handleError @ qtloader.js:258
VrmRemoteConsole2Modal.vue:280 Error: Could not fetch WebAssembly
dashboard:1 Mixed Content: The page at 'https://vrm.victronenergy.com/installation/133622/dashboard' was loaded over HTTPS, but requested an insecure resource 'http://133622-gui.proxyrelay3.victronenergy.com/auth/login.php?page=/gui-beta/venus-gui-v2.wasm'. This request has been blocked; the content must be served over HTTPS.
qtloader.js:258 Error: Could not fetch WebAssembly
handleError @ qtloader.js:258
```

mpvader, 2025-02-06:
In the msgs posted by Chris it says that it tries to load the wasm file over http. Which, on VRM it shouldn’t. Not for any security profile. It should load it from https always.

I don’t know where the logic is for using http or https, perhaps in one of the loader files?

AlexImb, 2025-02-06:
Discussed in Slack. Is an nginx config problem, @wiebeytec will fix.

wiebeytec, 2025-02-06:
Fix written, and offered for inclusion in a next 3.60 testing release.

chriadam, 2025-02-10:
TYVM.  OOI what is the precise number of the 3.60 release which will include it?

jhofstee, 2025-02-10:
It should be fixed in v3.60~25

closed 2025-02-10 by jhofstee