victronenergy/gui-v2 #1920: Introduce "QmlObject" to simplify "_private" properties

MikeTrahearn-Qinetic, 2025-02-19:
Proposing a new Victron.VenusOS / Core / QmlObject: 

A QtObject but with a default property alias to a property list<QtObject> - preferably in C++ as a [default] QQmlListProperty<QtObject>.
This would remove much of the “pseudo-private properties” and make the QML much more readable. It would also remove the ambiguity between accessing QtObject properties with its _propertyName or its id; only the id remains which causes a correct qualification which can be detected and validated by qmllint going forward.

chriadam, 2025-02-19:
Yeah, it's a good idea, solves having `property VeQuickItem _whatever: VeQuickItem { id: idWhatever }` etc

MikeTrahearn-Qinetic, 2025-02-19:
As a starting point we could add the class into the Core API. 
Then perhaps one or two example usages by way of demonstration linked to this ticket.
Then we can gradually clean up outside of this ticket as we go seeing as this would otherwise be an enormous task.

MikeTrahearn-Qinetic, 2025-02-19:
The following would be possible:

```
QmlObject {

   QtObject {
      id: heyImPrivateHaha
   }

   QtObject {
      id: soAmI
   }

   QmlObject {
      id: alsoImEvenNested

      QtObject {
         id: arentYouLucky
      }
   }
}
```

commit f0e1040c (2025-02-19): Add QmlObject for future simplification of "private" properties

Fixes #1920
commit a46b54fc (2025-02-19): Add QmlObject for future simplification of "private" properties

Fixes #1920
commit d26a50c2 (2025-02-21): Add QmlObject for future simplification of "private" properties

Fixes #1920
commit 92ee8787 (2025-02-21): Add QmlObject for future simplification of "private" properties

Fixes #1920