victronenergy/gui-v2 #1922: In Wasm, show the time according to the time zone setting

blammit, 2025-02-19:
The clock time in the status bar should showing the time according to the time zone set in the Date & Time settings, not the local time from the browser.

This aligns with the gui-v1 behaviour, where users would see the clock time according to the actual device when accessing it via Remote Console over VNC.

chriadam, 2025-02-27:
In Qt 6.9 proper timezone support will be added to Qt WebAssembly builds (see https://bugreports.qt.io/browse/QTBUG-68812 for info).

Until then, we have a couple of options:

1) venus-platform (or some other appropriate service on device) could publish the timezone offset data we need to perform the transform manually.  Currently, we have the following paths: `com.victronenergy.platform/Device/Time` contains the seconds since epoch (effectively, the output of `date +%s`), and `com.victronenergy.settings/System/TimeZone` contains a timezone name (e.g. "Antarctica/Macquarie").  Unfortunately, in WebAssembly, we don't have a valid tzdb so we cannot determine the appropriate offset for that given timezone name.  If we had some path like `com.victronenergy.settings/System/TimeZoneOffset` which gave an offset, then we could do the transformation manually (i.e. read platform/Device/Time and convert to a UTC datetime, then apply the settings/System/TimeZoneOffset value to get to the final timestamp for display).

2) we could bundle and use moment.js to perform the timezone transform.  See e.g. https://stackoverflow.com/questions/58490055/qt-to-qml-conversion-qdatetime-date for an example of this.  It means including another MIT-license file in the packaging, etc, and will bloat the size of the final binary a bit.

@ReinvdZee how difficult would it be to provide what we need for (1), i.e. for venus-platform to expose the value returned by
`env TZ="$(dbus-send --system --print-reply --reply-timeout=2000 --type=method_call --dest=com.victronenergy.settings /Settings/System/TimeZone com.victronenergy.BusItem.GetValue | tail -1 | sed 's/^[^"]*"\([^"]*\)".*/\1/')" bash -c 'date +%z'`
as `com.victronenergy.settings/System/TimeZoneOffset` ?

ReinvdZee, 2025-03-03:
@chriadam 

I'm fine with creating an additional path for this, something like: `com.victronenergy.platform/Device/TimeZoneOffset`. No need to create it in `com.victronenergy.settings` since its not a setting and will be recalculated on boot and every timezone change

And how would you like to have the offset to be formatted? In '+hhmm' format? Or will an integer corresponding to the number of seconds or minutes relative to UTC do? 

I'm planning on using `tm_gmtoff` from `struct tm`, which holds the offset in seconds.

chriadam, 2025-03-04:
That'd be perfect, thanks!

mpvader, 2025-04-03:
discussed, goes on hold until there is Qt 6.9