victronenergy/gui-v2 #1938: Improve visual performance of pushing new pages onto the stack view

blammit, 2025-02-27:
Some pages have a lot of items. On a GX device, there may be a noticeable delay when opening a page, if it takes time for the page to be created before it is pushed onto the stack view. 

For example, from the Device List:

1. Tap on the settings page for a vebus device
2. Notice there is a delay of a few hundred milliseconds
3. The new page object slides into view as it is pushed onto the stack

One to avoid the apparent lagginess is to:

1. Start creating the page object asynchronously in the background (from `PageStack.qml` where it calls `Qt.createComponent()`).
2. Push a page placeholder/container onto the stack and slide it into view. This should be animated on the render thread to avoid disrupting the objecting loading.
3. When the page object creation finishes, show it in the page container, and fade it into view.

MikeTrahearn-Qinetic, 2025-02-27:
Does this happen only on the first time - are we making any benefits from the .qmlc files if they are generated at all by the JIT compiler? Just wondering if we have a QML Profile of this particular case to see what is taking the time here. Perhaps async loading images for example could help too if it doesn't disrupt layout.

Just wondering at a high level and considering that the mechanism for #1071 is still being discussed, why we can't at least allow qmlcachegen to run and get the byte code for the QML into the resources already for the **shipped QML** code and then provide an ancillary mechanism for loading **custom QML**. Appreciating the comment about the GX device having a hard time if there are lots of non-object components declared in a QML file which are not yet created. Apologies for any lack of prior knowledge on this one, I'm just trying to understand why we are where we are and if any of the prior rationale has been removed.

chriadam, 2025-05-19:
Higher risk, as discussed in the meeting this should be delayed until after milestone 1.1.0.  Leaving current label for now, we can do a bulk migrate to 1.2.0 once we close the 1.1.0 label.

chriadam, 2026-08-20:
I've been making changes to address this, see https://github.com/victronenergy/gui-v2/pull/3150