victronenergy/gui-v2 #1944: Grid meter device page does not load on local GUI

ReinvdZee, 2025-02-27:
Seen on Venus OS 3.60~34 (**built with OE ScarthGap**) on the [ESS 3-phase system](https://vrm.victronenergy.com/installation/473156/dashboard).

The devicepage of the grid meter is empty on the local GUI, but only when opened from the device list:

[image]

When opening the same page using the widget from the overview page, it opens, although it takes some time.
On WASM, there are no issues.

blammit, 2025-03-13:
@ReinvdZee I'm not able to reproduce this on device with the grid simulation from `/opt/victronenergy/dbus-recorder/grid.csv`. The page loads as expected from both the Device List and Overview. Do you have any tips for reproducing it?

ReinvdZee, 2025-03-13:
@blammit I can also not reproduce it anymore on this system with 3.60~39. 
The issue seems to have resolved itself. Closing.

closed 2025-03-13 by ReinvdZee