victronenergy/gui-v2 #1947: Electric Propulsion UI (boat page) v1

DanielMcInnes, 2025-02-27:


DanielMcInnes, 2025-03-04:
Developer hand-off slides: https://docs.google.com/presentation/d/1NNTxaMKo4XhlhcXxbIy4TgXp5xo7xkXlDwg2MMb56AA/edit#slide=id.g339bc4d97d5_0_54
Figma link: https://www.figma.com/design/GjCtz3A3jqSGvyjVvOm6RZ/GX-Touch-design?node-id=10286-179939&t=lF6TbFrCd1AdPI2X-1
Link to venus-docker branch with simulated recordings: https://github.com/victronenergy/venus-docker/tree/philipptrenz/motordrive

DanielMcInnes, 2025-03-07:
Hi @philipptrenz , re. "Make max speed and RPM  configurable via “Minimum and maximum gauge ranges”", I think this is something that would be stored in the dbus path 'Settings/gui/gauges'. This will need platform changes to support the new path. I can work around it for the time being, but it will need to be added to the platform before this goes in for code review.

philipptrenz, 2025-03-07:
Hi @DanielMcInnes, 

To 1: Okay, can you send me a proposal of the necessary paths and its units? Then I'll arrange that they get added shortly.

To 2: Nice, this should be sufficient for now.

DanielMcInnes, 2025-03-07:
Hi @philipptrenz , that looks good, except for `...Rpm/Max`, which I think should have a `DC/Motordrive` prefix, as follows:
`/Settings/Gui/Gauges/DC/Motordrive/Rpm/Max				float`
Other gauges might have a max rpm value, eg generator or something else.

philipptrenz, 2025-03-07:
Hi @DanielMcInnes, I thought of them as more general settings, but I agree that at least RPM should be scoped to the motordrive. For speed, I think it's fine to have a global max value, or not? Here's my refined proposal:

```
/Settings/Gui/ElectricPropulsionUI/Enabled          uint
/Settings/Gui/Gauges/Motordrive/DC/Current/Max      float
/Settings/Gui/Gauges/Motordrive/DC/Power/Max        float
/Settings/Gui/Gauges/Motordrive/Rpm/Max             float
/Settings/Gui/Gauges/Speed/Max                      float
```

DanielMcInnes, 2025-03-07:
Hi @philipptrenz , this looks good to me. Yes, I think it's fine for speed to have a global max value.

DanielMcInnes, 2025-03-08:
Hi @philipptrenz , for the gear selection, we show "F N R" for forward, neutral and reverse. Would you expect these letters to be translated?

philipptrenz, 2025-03-08:
Hi @DanielMcInnes, also a good one. I guess having them translated won't be well understood, let's keep them untranslated.

DanielMcInnes, 2025-03-12:
Hi @philipptrenz , in https://docs.google.com/presentation/d/1NNTxaMKo4XhlhcXxbIy4TgXp5xo7xkXlDwg2MMb56AA/edit#slide=id.g2bad103b4a7_1_1319 it says:
`Show with “-” on current draw, without “-” if battery is charging / motor recuperates (For uniformity throughout the system)`

So, if the motordrive was drawing 312A, the display would show `-312A`, would the value at `[system_battery_service]/Dc/0/Current` be positive or negative?

philipptrenz, 2025-03-12:
Hi @DanielMcInnes,

`com.victronenergy.motordrive/Dc/0/Current` is positive on power draw, so you have to multiply by `-1`. When using the `[system_battery_service]/Dc/0/Current`, it's negative on power draw, so please use it as is.

philipptrenz, 2025-03-12:
Additionally: @ReinvdZee is working on the settings paths and had a good point: `/Settings/Gui/Gauges/DC/Motordrive/Current/Max` and `/Settings/Gui/Gauges/DC/Motordrive/Power/Max` should reference the same. Therefore, we'd like to just have `/Settings/Gui/Gauges/DC/Motordrive/Current/Max` and gui-v2 to calculate the power from it, if necessary. Any objections?

DanielMcInnes, 2025-03-12:
> Additionally: [@ReinvdZee](https://github.com/ReinvdZee) is working on the settings paths and had a good point: `/Settings/Gui/Gauges/DC/Motordrive/Current/Max` and [...]

Hi @philipptrenz , I will need to multiply the current by some nominal voltage to calculate the maximum power. Is that voltage available somewhere? I don't think I can use `com.victronenergy.motordrive/Dc/0/Voltage/`, as that value changes over time, and would give me changing values for `.../Power/Max`

DanielMcInnes, 2025-03-12:
Hi @philipptrenz , re. using battery power when motordrive power is unavailable:

[image]

it occurs to me that we might have an electric boat with a generator providing power to the battery, which is in turn providing power to the motordrive. In this situation, the net power to/from the battery would be offset by the generator power, which would give us misleading results. Does this sound right, or have I misunderstood?

DanielMcInnes, 2025-03-13:
Hi @philipptrenz , what does this setting do?

[image]

Does it change the "14" in this image to eg. "1hr 30m"?

[image]

If so, I think we will need another setting, something like: 
`/Settings/Gui/ElectricPropulsionUI/CentreGauge/Type              uint (0|1) #0: speed 1: time to go`

DanielMcInnes, 2025-03-13:
Hi @philipptrenz , the settings design seems to imply that we should display only 1 of motor temperature, controller temperature, and coolant temperature.

[image]

 The boat page design implies that we display all of them (when available). 

[image]

Can you clarify please?

DanielMcInnes, 2025-03-13:
Hi @philipptrenz , just confirming - do you want to be able to set light & dark mode for the boat page independently of the rest of the gui?

[image]

DanielMcInnes, 2025-03-13:
Hi @philipptrenz , last question I think. 'Boat Length' is in the settings, what is it used for?

chriadam, 2025-03-13:
> Hi [@philipptrenz](https://github.com/philipptrenz) , just confirming - do you want to be able to set light & dark mode for the boat page independently of the rest of the gui?

I think this is not a good idea, personally.

chriadam, 2025-03-13:
> Hi [@philipptrenz](https://github.com/philipptrenz) , last question I think. 'Boat Length' is in the settings, what is it used for?

This is used to calculate the optimal speed for the boat to maximise efficiency (it's based on skin-friction and the bow-wave produced, if memory serves).

philipptrenz, 2025-03-13:
Hi @DanielMcInnes, like just discussed, please ignore the design for the custom settings, which should make [this](https://github.com/victronenergy/gui-v2/issues/1947#issuecomment-2719696895), [this](https://github.com/victronenergy/gui-v2/issues/1947#issuecomment-2719713284), [this](https://github.com/victronenergy/gui-v2/issues/1947#issuecomment-2719714738) and [this](https://github.com/victronenergy/gui-v2/issues/1947#issuecomment-2719725253) get irrelevant.

philipptrenz, 2025-03-13:
> Hi [@philipptrenz](https://github.com/philipptrenz) , re. using battery power when motordrive power is unavailable:
>
> [image]
>
> it occurs to me that we might have an electric boat with a generator [...]

Regarding this one: You're absolutely right, the right way would be to show the "DC load", as it is shown in the Overview, andn not the battery. Luckily, this is in line with what we have just discussed regarding positive/negative values. So we adjust the logic as follows:

- For a motordrive, consumption is positive, potential production is negative (as reported by the motordrive service)
- If there is no motordrive, we will not show the `[system_battery_service]` current/power, but the "DC load" power/current, which follows the same behaviour as above (consumption=positive, production=negative)

So, we don't need any multiplication with `-1` and can use the values as given. I think the "DC load" value comes from dbus-systemcalc, but I guess you're quicker to find out then I am 😉

philipptrenz, 2025-03-13:
> > Additionally: [@ReinvdZee](https://github.com/ReinvdZee) is working on the settings paths and had a good point: `/Settings/Gui/Gauges/DC/Motordrive/Current/Max` and `/Settings/Gui/Gauges/DC/Motordrive/Power/Max` should reference the same. Therefore, we'd like to just have `/Settings/Gui/Gauges/DC/Motordrive/Current/Max` and gui-v2 to calculate the power from it, if necessary. Any objections?
> 
> Hi [@philipptrenz](https://github.com/philipptrenz) , I will need to multiply the current by some nominal voltage to calculate the maximum power. Is that voltage available somewhere? I don't think I can use `com.victronenergy.motordrive/Dc/0/Voltage/`, as that value changes over time, and would give me changing values for maximum power.

You're right with that, it would slightly fluctuate, but that's actually just mimicking the physics and therefore should be fine. Or is it a technical challenge?

Please also note: I'm still in discussion with Rein about the settings paths and we might use `/Settings/Gui/Gauges/Motordrive/DC/Power/Max` instead of `/Settings/Gui/Gauges/Motordrive/DC/Current/Max`, as adding the max power rating of the motordrive will be easier for the user. So instead, you would need to divide through the voltage. I'll let you know the full set of paths with their correct spelling as soon as it is decided. Sorry for the back and forth on this ...

philipptrenz, 2025-03-13:
Hi @DanielMcInnes, 

here are the paths that Rein added, also see commit referenced above:

```
/Settings/Gui/ElectricPropulsionUI/Enabled
/Settings/Gui/Gauges/MotorDrive/Power/Max
/Settings/Gui/Gauges/MotorDrive/Rpm/Max
/Settings/Gui/Gauges/Speed/Max
```

It should get included in Venus within the next week or so.

philipptrenz, 2025-03-17:
> Hi [@philipptrenz](https://github.com/philipptrenz) , I will need to multiply the current by some nominal voltage to calculate the maximum power. Is that voltage available somewhere? I don't think I [...]

Hi @DanielMcInnes, Rein added `/MotorDrive/Voltage` to systemcalc, so that you can take it from there to calculate the current.

philipptrenz, 2025-03-18:
Hi @DanielMcInnes, the systemcalc settings are now included in Venus OS since v3.60~44.

DanielMcInnes, 2025-03-24:
Hi @philipptrenz , should we have a settings page for `/Settings/Gui/ElectricPropulsionUI/Enabled`, or is that meant to be set via VictronConnect?

philipptrenz, 2025-03-24:
One general thing of the "Minimum and maximum gauge ranges" page. While "Auto-ranging" is enabled, shouldn't all these entries be non-editable? Maybe you can quickly confirm @ReinvdZee ?

commit 6f89c5d8 (2025-03-12): Implement the electric propulsion UI (boat page)

Fixes #1947
commit 5cf18fd1 (2025-03-12): Implement the electric propulsion UI (boat page)

Fixes #1947
commit b122da3d (2025-03-12): Implement the electric propulsion UI (boat page)

Fixes #1947
commit e2990df2 (2025-03-12): Implement the electric propulsion UI (boat page)

Fixes #1947
closed 2025-04-24 by DanielMcInnes