victronenergy/gui-v2 #1950: Brief center gauges show gauge types that weren't selected in settings

blammit, 2025-03-02:
In Settings > General > Display & Appearance > Brief view levels, if you set "Level" to a particular tank type, but there are no tanks of that type on the system, then the Brief view shows another available tank type instead.

This is confusing for the user as this means the Brief page shows a type that wasn't actually selected in the settings.

blammit, 2025-03-02:
The reason for this is that the default `/Settings/Gui/BriefView/Level/<0-3>` values are [Battery, Fuel, Fresh water, Black water]. If the user does not go to the settings and customize the Brief gauges, we still want to show their tanks on the Brief page. 

The default levels should have been initially put into the gui-v2 code, rather than into the settings values, so that gui-v2 could detect whether the settings had been changed by the user.

Perhaps this can be resolved by #1949 which requires a new set of Brief level settings of `string` type. If all the values are initially empty strings, then gui-v2 can detect whether they have been changed by the user.

chriadam, 2025-03-03:
Is this really for milestone-1.0.0?  I assume it's meant to be 1.1.0?

blammit, 2025-05-01:
This has been resolved as per the referenced commits.

closed 2025-05-01 by blammit