victronenergy/gui-v2 #1961: Brief page (local/remote) does not show Grid / Shore connection

baudje, 2025-03-09:
The shore connection is shown as not active on the Brief page. On the overview page it is shown.

[image]

[image]

OS 3.54 but also on latest beta.

blammit, 2025-03-12:
@baudje could you please add victron.vrm (at) qinetic.com.au as a user to your installation so we can investigate/verify via VRM?

chriadam, 2025-03-13:
Is it the case that we should preferentially show the generator over shore if the generator is running, but we should show shore over generator if the generator is stopped?  @jepefe perhaps could comment on this

chriadam, 2025-03-13:
(In terms of implementation, I suspect we would need to update the logic in AcInputs.qml for the highlightedInput property, to change which input is "highlighted" depending on the operational state of the generator device.)

baudje, 2025-03-13:
> [@baudje](https://github.com/baudje) could you please add victron.vrm (at) qinetic.com.au as a user to your installation so we can investigate/verify via VRM?

let me know if it does not work

kwindrem, 2025-03-13:
I think the logic is fairly simple and can be done without additional info from the Inverter:

Determine the current AC input, then see if it is a generator.
If it is the generator, give preference to the generator tile, if not, give preference to the grid tile.
It should not matter if the generator is running and in fact the generator may be running for test, started manually and/or started externally WITHOUT it feeding the inverter. Giving preference to the generator tile in this case would not be appropriate since the grid would still be supplying power to the interior.

blammit, 2025-03-13:
> > [@baudje](https://github.com/baudje) could you please add (-) as a user to your installation so we can investigate/verify via VRM?
> 
> let me know if you have issues logging in

Connected successfully, thanks!

commit e186b650 (2025-03-13): Brief: show correct AC input if both inputs are on vebus or acsystem

When determining the highlightedInput (i.e. the AC input shown on the
Brief page), first check /Ac/ActiveIn/Source against each input's
source, then /Ac/ActiveIn/ServiceType against each input's type.

The previous code was incorrect as it checked /Source or /ServiceType
against the first input, then then /Source or /ServiceType against the
second input. This meant if both inputs were on vebus, then AC-in 1
could be returned even if its source did not match the active source,
because its service type still matched the active [... cut]
PR #1978 (2025-03-17): Brief: show correct AC input if both inputs are on vebus or acsystem
When determining the highlightedInput (i.e. the AC input shown on the Brief page), first check /Ac/ActiveIn/Source against each input's source, then /Ac/ActiveIn/ServiceType against each input's type.

The previous code was incorrect as it checked /Source or /ServiceType against the first input, then then /Source or /ServiceType against the second input. This meant if both inputs were on vebus, then AC-in 1 could be returned even if its source did not match the active source, because its service type still matched the active  [... cut]
commit ebadcba9 (2025-03-17): Brief: show correct AC input if both inputs are on vebus or acsystem

When determining the highlightedInput (i.e. the AC input shown on the
Brief page), first check /Ac/ActiveIn/Source against each input's
source, then /Ac/ActiveIn/ServiceType against each input's type.

The previous code was incorrect as it checked /Source or /ServiceType
against the first input, then then /Source or /ServiceType against the
second input. This meant if both inputs were on vebus, then AC-in 1
could be returned even if its source did not match the active source,
because its service type still matched the active [... cut]
closed 2025-03-17 by blammit