victronenergy/gui-v2 #1964: Unable to open any settings pages if access level is "User"

blammit, 2025-03-11:
Go to settings and change the access level to "User". Observe that now if you click on any settings menu items, the "Setting locked for access level" toast is shown and the page cannot be opened.

blammit, 2025-03-11:
This is related to the removal of `enabled: userHasReadAccess` in ListNavigation for #1818. We need some alternative way of allowing ListNavigation to be clicked; it should be clickable if the user has read access for the item.

MikeTrahearn-Qinetic, 2025-03-12:
c.f ListItem code in its PressArea:

```
        onClicked: {
            if (root.interactive) {
                if (!root.userHasWriteAccess) {
                    pressArea.toast?.close(true) // close immediately
                    //% "Setting locked for access level"
                    pressArea.toast = Global.notificationLayer.showToastNotification(VenusOS.Notification_Info, qsTrId("listItem_no_access"))
                } else {
                    root.clicked()
                }
            }
        }
```
ListNavigation components have `interactive: true` but this isn't enough to get past the inner "if" block in ListItem.

MikeTrahearn-Qinetic, 2025-03-12:
For reference ListNavigation used to have `enabled: userHasReadAccess` before this was changed to `interactive: true`

commit 0ebe7bea (2025-03-12): Fix being unable to open any settings pages if access level is "User"

Fixes #1964
commit 9ad98512 (2025-03-12): Fix being unable to open any settings pages if access level is "User"

Fixes #1964
PR #1968 (2025-03-13): Fix being unable to open any settings pages if access level is "User"
Fixes #1964
commit bee0d974 (2025-03-13): Fix being unable to open any settings pages if access level is "User"

Fixes #1964