victronenergy/gui-v2 #1991: "Only" 5 batteries visible under devices

mr-manuel, 2025-03-18:
Hello,

it might be a very edge case, but only 5 out of 14 batteries are visible under devices. In GUIv1 all are visible. It could also be a MQTT issue, that some data is missing.

[image]

[image]

[image]

blammit, 2025-03-18:
Only the batteries in com.victronenergy.system/Batteries will be shown in this battery list.

If you go to Settings > System setup > Batteries > Battery measurements, you can select the batteries to be added to this list. Does that work for you?

mr-manuel, 2025-03-19:
>  Settings > System setup > Batteries > Battery measurements

This should never influence the effective device list, but at least the batteries are in the list. If I open the entry only the batteries that also are available in the devices page have all values, the other do not.

[image]

[image]

[image]

```
root@einstein:~# dbus -y com.victronenergy.system /AvailableBatteryMeasurements GetValue
{'com_victronenergy_battery_10/Dc/0': 'SerialBattery(Daly CAN) on CAN '
                                      'socketcan:can0__10',
 'com_victronenergy_battery_11/Dc/0': 'SerialBattery(Daly CAN) on CAN '
                                      'socketcan:can0__11',
 'com_victronenergy_battery_12/Dc/0': 'SerialBattery(Daly CAN) on CAN '
                                      'socketcan:can0__12',
 'com_victronenergy_battery_13/Dc/0': 'SerialBattery(Daly CAN) on CAN '
                                      'socketcan:can0__13',
 'com_victronenergy_battery_14/Dc/0': 'SerialBattery(Daly CAN) on CAN '
                                      'socketcan:can0__14',
 'com_victronenergy_battery_3/Dc/0': 'SerialBattery(Daly CAN) on CAN '
                                     'socketcan:can0__3',
 'com_victronenergy_battery_4/Dc/0': 'SerialBattery(Daly CAN) on CAN '
                                     'socketcan:can0__4',
 'com_victronenergy_battery_5/Dc/0': 'SerialBattery(Daly CAN) on CAN '
                                     'socketcan:can0__5',
 'com_victronenergy_battery_6/Dc/0': 'SerialBattery(Daly CAN) on CAN '
                                     'socketcan:can0__6',
 'com_victronenergy_battery_7/Dc/0': 'SerialBattery(Daly CAN) on CAN '
                                     'socketcan:can0__7',
 'com_victronenergy_battery_8/Dc/0': 'SerialBattery(Daly CAN) on CAN '
                                     'socketcan:can0__8',
 'com_victronenergy_battery_9/Dc/0': 'SerialBattery(Daly CAN) on CAN '
                                     'socketcan:can0__9',
 'default': 'Automatic',
 'nobattery': 'No battery monitor'}

root@einstein:~# dbus -y com.victronenergy.system /AvailableBatteryServices GetValue
('{"default": "Automatic", "nobattery": "No battery monitor", '
 '"com.victronenergy.battery/3": "SerialBattery(Daly CAN) on CAN '
 'socketcan:can0__3", "com.victronenergy.battery/4": "SerialBattery(Daly CAN) '
 'on CAN socketcan:can0__4", "com.victronenergy.battery/5": '
 '"SerialBattery(Daly CAN) on CAN socketcan:can0__5", '
 '"com.victronenergy.battery/6": "SerialBattery(Daly CAN) on CAN '
 'socketcan:can0__6", "com.victronenergy.battery/7": "SerialBattery(Daly CAN) '
 'on CAN socketcan:can0__7", "com.victronenergy.battery/8": '
 '"SerialBattery(Daly CAN) on CAN socketcan:can0__8", '
 '"com.victronenergy.battery/9": "SerialBattery(Daly CAN) on CAN '
 'socketcan:can0__9", "com.victronenergy.battery/10": "SerialBattery(Daly CAN) '
 'on CAN socketcan:can0__10", "com.victronenergy.battery/11": '
 '"SerialBattery(Daly CAN) on CAN socketcan:can0__11", '
 '"com.victronenergy.battery/12": "SerialBattery(Daly CAN) on CAN '
 'socketcan:can0__12", "com.victronenergy.battery/13": "SerialBattery(Daly '
 'CAN) on CAN socketcan:can0__13", "com.victronenergy.battery/14": '
 '"SerialBattery(Daly CAN) on CAN socketcan:can0__14"}')

root@einstein:~# dbus -y com.victronenergy.system /AvailableBmsServices GetValue
[{'instance': 3, 'name': 'Daly #3'},
 {'instance': 4, 'name': 'Daly #4'},
 {'instance': 5, 'name': 'Daly #5'},
 {'instance': 6, 'name': 'Daly #6'},
 {'instance': 7, 'name': 'Daly #7'},
 {'instance': 8, 'name': 'Daly #8'},
 {'instance': 9, 'name': 'Daly #9'},
 {'instance': 10, 'name': 'Daly #10'},
 {'instance': 11, 'name': 'Daly #11'},
 {'instance': 12, 'name': 'Daly #12'},
 {'instance': 13, 'name': 'Daly #13'},
[... 101 lines cut ...]
  'state': 0,
  'voltage': 53.5},
 {'active_battery_service': False,
  'bmsstate': 9,
  'current': -0.0,
  'id': 'com.victronenergy.battery.can0__0x0d',
  'instance': 13,
  'name': 'Daly #13',
  'power': -0.0,
  'soc': 45.3,
  'state': 0,
  'voltage': 53.5},
 {'active_battery_service': False,
  'bmsstate': 9,
  'current': -0.0,
  'id': 'com.victronenergy.battery.can0__0x0e',
  'instance': 14,
  'name': 'Daly #14',
  'power': -0.0,
  'soc': 47.4,
  'state': 0,
  'voltage': 53.5}]
```

@mpvader is there a battery limit in Venus OS? Should this further be investigated, even if this is probably a very rare case?

blammit, 2025-05-01:
> This should never influence the effective device list, but at least the batteries are in the list. If I open the entry only the batteries that also are available in the devices page have all values, the other do not.

I see, I misunderstood. Yes, all `battery` services should definitely appear in the Device List. There aren't any limitations built into gui-v2 to restrict the displayed battery services in the Device List, so there could be some limitation with what's being reported via MQTT, or a bug in the model. Would need to check with a test installation.

WaldemarFech, 2025-06-03:
Hey everyone, I’d like to add my findings.
Here’s a quick rundown of my current test setup—no idea yet whether it’s directly related to the issue:

---

### Test setup

* **4 × JKBMS\_PB** connected in parallel over one USB-RS485 adapter

  * Modbus addresses: `0x01`, `0x02`, `0x03`, `0x04`
  * All four devices are detected.

### What I’m seeing

1. **Settings → Devices**

   * Only **three** batteries show up (see screenshot).
   * Every now and then the entry for Battery 1 flashes briefly at the bottom.

2. **Settings → System Setup → Batteries**

   * All **four** batteries are listed and their values update live.
   * **BUT:** When I tap the missing JK entry, the detail view is empty (photo attached).
     The other three open fine.

3. **Overview → Batteries**

   * Again all four batteries are visible.
   * Same problem: selecting **JK 1** shows a blank screen.

### Additional context

* I still have **two older JKBMS (non-PB)** running.

  * When those are active together with the 4 PB units (so 6 batteries total), the code crashes—different story, will open a separate issue or maybe send a PR soon.
  * For now I’ve patched it so the system stays up, but once both JKBMS and JKBMS\_PB are active the whole stack freezes and eventually reboots (watchdog).

---

### Next step I’m trying

I’m about to test **6 × USB-RS485 adapters** on a USB hub, i.e. regular RS-485 without device addresses. Hope I can get this running before I head to work.

[image]

[image]

[image]

[image]

edit:
root@einstein:~# dbus -y | grep com.victronenergy.battery | sort
com.victronenergy.battery.aggregate
com.victronenergy.battery.ttyUSB1__0x01
com.victronenergy.battery.ttyUSB1__0x02
com.victronenergy.battery.ttyUSB1__0x03
com.victronenergy.battery.ttyUSB1__0x04

mpvader, 2025-06-05:
Discussed in the meeting today with Manuel:

- its not easy to reproduce.
- if Manuel sees it again on a system he'll investigate more.
- make sure to distinguish between an issue in systemcalc, not listing available batteries, versus an issue in gui-v2, not listing them in the device list.