victronenergy/gui-v2 #1993: Improve Security Profile settings UI flow

chriadam, 2025-03-20:
Philipp has updated the design to separate the password and profile settings.
(Double check that the password setting is indeed a separate path, and isn't stored as a JSON array value in the profile setting.)

MikeTrahearn-Qinetic, 2025-03-24:
The current password validation is buried into the ListRadioButtonGroup's RadioListButtonPage's GradientListView ListRadioButton delegate additional child SettingsColumn. This is used in three places:

1) Changing the Access Level
2) Changing the Security Profile (to be removed by this ticket)
3) In the Settings Demo Page.

Due to 1) we must leave all the existing in-line password handling in place. However 2) will require more thought since we explicitly do NOT want to use the in-line password (from the **promptPassword: true** optionGroup key value) and move this into a Dialog instead.

MikeTrahearn-Qinetic, 2025-03-24:
Vertically centering the RadioButton's rings as proposed will be problematic; the RadioButton exists on the top of a vertical stack of varying content including captions and password columns. The only way to fix this thoroughly is redesign the layout of the ListRadioButton.

chriadam, 2025-05-05:
Discussed with Philipp that since there is no intention to change the API, we really just need to add a button to allow the user to easily update just the password (without changing the profile setting).  The PR https://github.com/victronenergy/gui-v2/pull/2097 adds this.

chriadam, 2025-05-07:
Merged to main, closing as fixed.

commit b8cadc4d (2025-03-31): New Security Profile Selection UX flow

Fixes #1993
commit 751240d8 (2025-03-31): New Security Profile Selection UX flow

Fixes #1993
commit ec43b313 (2025-04-02): New Security Profile Selection UX flow

Fixes #1993
commit 2bc9c5cf (2025-04-02): New Security Profile Selection UX flow

Fixes #1993
closed 2025-05-07 by chriadam