victronenergy/gui-v2 #1995: DVCC: Message that tells user to do a VE.Bus reset is also shown in systems without a VE.Bus inverter (Multi RS or HS19)

izak, 2025-03-21:
[image]

Also see https://github.com/victronenergy/venus-private/issues/479

You can use com.victronenery.system/Devices/NumberOfVebusDevices to see if there is a VE.Bus inverter present.

chriadam, 2025-04-23:
Created speculative fix PR above.  Again, not sure how to test this myself.  @ReinvdZee perhaps could help with this?

ReinvdZee, 2025-04-23:
@chriadam I quickly tested your branch on a system with vebus inverter and on a system with Multi RS. Upon disabling DVCC, the toast still shows up on systems with a vebus inverter and it does not show up on a system with Multi RS.

So this works as expected.