victronenergy/gui-v2 #1998: VE.Bus: Add confirmation pop up to "Resart VE.Bus system" button

vejpasop, 2025-03-24:
Improvement request.

Currently it is too easy to restart the Ve,Bus system by mistake. Especially while navigating on a touch screen. A simple touch of the restart button immediately results in a reset of  Multi's/Quattro's (along with a loss of power on the output). 
I would recommend adding "are you sure yes/no" confirmation popup.

[image]

chriadam, 2025-03-25:
Simple to implement, and an important UI hygiene issue.  Setting to high_prio and marking for 1.1.0.

chriadam, 2025-04-04:
Created PR https://github.com/victronenergy/gui-v2/pull/2030 for this; any testing would be appreciated!

chriadam, 2025-04-07:
Merged, closing as fixed.

closed 2025-04-07 by chriadam

vejpasop, 2025-04-07:
It does. The popup states "Restarting the VE.Bus system will reset any inverter on the bus, and result in a loss of power to their outputs."