victronenergy/gui-v2 #2001: Victron Cerbo GX Screen Goes Black / screen is frozen after upgrading to GUI-v2

marekchowaniok, 2025-03-25:
Hi, 

I think there might be few people experiencing problem described here: 

https://community.victronenergy.com/t/victron-cerbo-gx-screen-goes-black-screen-any-fixes/23502

For easier reading I will put here the details: 

user: cowem
I am experiencing an issue where my Victron Cerbo GX screen intermittently goes black screen but the system continues running in the background. Restarting sometimes helps, but the problem returns.

user: marek1 (myself)
I am experiencing similar / same issue. I think it might be connected with latest firmware. My cerbo screen gets dark ( I think this is standard behavior) but when I wake up screen by touching it I can see that it is frozen and numbers doen’t change. It looks like it has frozen.
My Cerbo is on v3.54.

You can read the thread but I found solution to this - reverting to gui-v1 :( 

user: marek1
hi lxonline

so I have solution for my specific situation.

I have update Cerbo to Beta version (something 3.60.xxxxx) - same result (screen frozen after while)
I have update Cerbo to prev version 3.52 - same result (screen frozen after while)

I have put it back to latest version 3.54 and I have changed DISPLAY to v1 (old classic view).
and it is working fine.

So there is something with new UI “http:/xxxxxx/gui-v2” which freezes the screen after while, but when it is put back to classic view it works correctly

Are you aware of this issue ? Someuser suggest not enough CPU or RAM ? Can I somehow check that ? 
Is there anything I can do to run on gui-v2 ? 

thanks

chriadam, 2025-03-25:
Can you ssh to the device?  If so, can you look to see whether `ps | grep gui-v2` shows that gui-v2 is still running or not?  If so, can you try `top` to see if gui-v2 is taking a lot of CPU time or not?

From what you describe, it sounds like either gui-v2 is "hung" (the above steps should let us know if that's the case), or that gui-v2 crashed and was not restarted automatically by the gui system service via /usr/bin/start-gui (but I don't understand what could lead to this case).

marekchowaniok, 2025-03-25:
I have turned back gui-v2 in my Cerbo display. 
I can see, that it takes around constantly 40% CPU. 

I will be monitoring it and wait till the screen freezes and update you how it looks like when it does so. 

[image]

marekchowaniok, 2025-03-25:
once the screen goes to sleep (dark) CPU usage drops down. 

[image]

But still I am able to wake the screen - will wait longer and provide feedback.

marekchowaniok, 2025-03-26:
Hi @chriadam 
today morning I went to check the Cerbo and I was able to replicate the issue. 
I touched the screen (it was black - on screensaver) - I could see that it was not taking CPU almost at all (around 5%) and screen wakes up but numbers freezes (it kind of jumps between 2 numbers). 

https://github.com/user-attachments/assets/99e283e2-6e6b-48a2-9f8e-3346c2de8253

When I access Cerbo through url/website everything works. 

Currently (screen woke up) CPU shows cca 40%

[image]

chriadam, 2025-03-27:
Extremely interesting.  I will try to do some investigation, although I have not been able to reproduce anything like this before.

chriadam, 2025-03-27:
After setting the Display Off Time to 10 seconds (via Settings / General / Display & Appearance / Display Off Time) and doing a bunch of testing (i.e. from Overview page, waiting 10 seconds to blank, waiting a few seconds more, pressing the screen, checking that the UI remains responsive) I cannot reproduce any issue.

I have tried with various start pages (Overview page, Brief page, Brief page with Side Panel open, etc).
I have tried waiting various lengths of time between blanking and waking device.
I have tried using the manual "snooze" button in the status bar to trigger blanking, versus waiting for the display off timeout.

In no case have I seen either of the issues you mentioned (either (1) that the screen cannot be woken back up, (2) that the screen wakes up but is "trapped" in a weird oscillating value page and cannot be navigated).

CPU usage of 40% by gui-v2 on the Overview page is high (and we do need to improve this in future) but not entirely unexpected - it renders a lot of individual items with a lot of transparency etc.  CPU usage of less than 5% when the screen is blanked is expected (when the screen is blanked, gui-v2 will be running and updating data, but NOT rendering; and it is rendering which takes the most CPU usage).  So I don't think this is an issue with CPU usage, per se.

The "flickering" might be an accurate depiction of the data in the system (e.g. could you do a `dbus-monitor --system` to see if something (systemcalc perhaps) is entering some sort of loop and publishing the same values over and over?) OR it might be the two most recent render buffers being swapped (but never drawn into) i.e. a problem with the graphics driver or Qt's usage of it, perhaps...

Can you confirm that there are NO modifications to your device (no patches, custom software, or other filesystem changes, etc)?

Which precise version did you start to see the issue?  Were you using gui-v2 prior to that update (i.e. did gui-v2 seem to gain this bug due to an update) or were you always using the older gui (v1) prior to the update in which you began seeing the problem?

marekchowaniok, 2025-04-01:
hi @chriadam 

I tried "dbus-monitor --system" there are many values changing on the screen - and not sure what I am supposed to search for. 

Are you working for Victron ? If I gave you my ID can you login and check in my system ? 

To your questions: 
* I have enabled ssh access 
* I have used "DESS hack" - it rewrites and put extra loging into Cerbo screen - to show what system is doing - but this should not be related because I had this flickering issue even before

except those 2 there were no modifications.

marekchowaniok, 2025-04-01:
just checked processes and I can see this:  (there is mentioned "systemcalc" ) in the processes

[image]

chriadam, 2025-04-03:
> I tried "dbus-monitor --system" there are many values changing on the screen - and not sure what I am supposed to search for.

Just wondering if the values you're seeing "flickering" are being provided by the system, or not.

Most likely, it isn't - I think it's far more likely to be a rendering issue - either in Qt or in the driver or in the hardware.
We are discussing internally and trying to figure out the path forward.  Thanks!

marekchowaniok, 2025-04-03:
If you guys need any help - like providing access or anything else let me know. 

I have returned it back to gui-v1 (classical view) in Cerbo - which works fine and in browser I use newUI (by adding "gui-v2" in url) so I got maximum from both worlds. 
But frankly I really like new version in Cerbo as well :)