victronenergy/gui-v2 #2002: Wrong detail page for AC charger widget

mebitek, 2025-03-25:
I do not know if it has already been fixed but in version 1.0.9 (venus OS 3.54) if you click on the AC charger widget (a shunt used as energy meter, type "ac charger" no info are displayed

[image]

[image]

the landing page seems to be the pageAcInModel one (displays just "AC totals" that is not related with the ac charger meter

blammit, 2025-03-27:
Might be able to fix this just by returning `PageAcCharger.qml` for the `DcInputWidget` `detailUrl` in the case where `inputType == VenusOS.DcInputs_InputType_AcCharger`.

blammit, 2025-03-27:
@mebitek can you please add victron.vrm (at) qinetic.com.au to your VRM installation so we can investigate/verify this?

mebitek, 2025-03-27:
@blammit I installed 1.1.7 and the issue is gone.
how to add the address to my installation?

blammit, 2025-03-28:
That's great news 👍 I will close this issue, then.

For future reference, if you go to the page for your installation in https://vrm.victronenergy.com, then click Settings > Users in the side bar, you can add users to the installation.

closed 2025-03-28 by blammit