victronenergy/gui-v2 #2006: Battery Page Click Not Responding in GUI-v2 WebAssembly Build (WASM) Description:

JigneshMali, 2025-03-27:
I am working on adding a new feature to gui-v2 and using GitHub Actions to build the WebAssembly (WASM) version.

The build completes successfully, installs correctly, and runs in the browser. However, when I click on the Battery Page in the menu, nothing happens. The click appears to be null (unresponsive), and navigation does not occur as expected.

This issue is specific to the WebAssembly build.

[image]

When I run gui-v2 on my Cerbo GX device via HDMI, the changes in my QML page work correctly.

However, when accessing the Remote Console via WebAssembly, the Battery Page click is unresponsive.

**Steps to Reproduce:**
Modify the QML files in gui-v2 to add a new feature.
Use GitHub Actions to build gui-v2 for WebAssembly following the official instructions.
Deploy and access the WebAssembly build in a browser.
Click on the Battery Page in the menu.
The page does not load or navigate; the click appears to be null.

**Expected Behavior:**
Clicking on the Battery Page should navigate to the battery information screen.
**but this issue** -> Clicking on the Battery Page results in no action (null click).

**Troubleshooting Done:**
✔️ Verified that the issue occurs in multiple browsers (Chrome, Firefox).
✔️ Checked browser console logs (please provide logs if available).
✔️ Compared WebAssembly build with local HDMI display (works fine on HDMI).
✔️ Ensured the QML file changes were correctly included in the build.
✔️ Verified that the WASM build is installed in the correct path.

**Questions:**
1.Are there any additional steps I might have missed when making QML changes and building gui-v2 for WebAssembly?
2.Does the Battery Page require any specific WebAssembly-related dependencies to enable navigation?
3.Could this be related to the QML Loader, page routing, or WebAssembly compatibility issues?

Any debugging tips to trace this issue in WebAssembly builds?

**Build Method:** GitHub Actions

Additional Notes:
This issue is only affecting WebAssembly builds, and the same QML changes work fine on a Cerbo GX HDMI display.

Would appreciate any insights or suggestions!

chriadam, 2025-03-27:
Just double checking: how do you install the outputs from the github action?  And precisely which outputs do you install?

I usually install the following files:
```
index.html  venus-gui-v2.js  venus-gui-v2.wasm.gz  victronenergy.svg  LICENSE.txt  qtloader.js  venus-gui-v2.wasm.sha256
```
to the directory:
```
/var/www/venus/gui-v2/
```
on the CerboGX device.

I then reboot the device (this ensures: 1) that nginx will re-resolve its cache, for local network connections; 2) that the script which detects changes to the gui-v2 blobs and reports those up to VRM).  If you are connecting to the device via the remote console via VRM, you will need to wait 10 minutes or so after the device has rebooted, for VRM to pull the updated blob from the device up into its cache.

JigneshMali, 2025-03-27:
Yes, I do the Same ->

`index.html  venus-gui-v2.js  venus-gui-v2.wasm.gz  victronenergy.svg  LICENSE.txt  qtloader.js  venus-gui-v2.wasm.sha256` 

to the directory :
`/var/www/venus/gui-v2/`

chriadam, 2025-03-27:
Then I'm not sure.  I can only suggest: open the Developer Tools / Browser Console and check to see if any warnings are printed when you load the page or click the button you have added, to see if there's some Reference Error or other issue.

JigneshMali, 2025-03-28:
1. I am using this for building wasm. -> [Building for WebAssembly via GitHub Actions](https://github.com/victronenergy/gui-v2/wiki/How-to-build-venus-gui-v2#building-for-webassembly-via-github-actions)
2. I think need help with more detailed instruction from bottom to top , so I can differentiate the steps that I may have missed. howere the using above instructions wasm build is successful and webconsole works. but it just does not let me use the battery page when I click it from devicelist. 
3. My changes in batterypages are working fine in local display gui-v2. it is working fine.

chriadam, 2025-03-28:
Is it the case also for a webassembly build from "main" branch?
If not, then it must be a regression caused by your code changes (and I can only suggest to look at the browser's javascript developer console, to check for reference errors or similar).

Note that just because something works in a device build (ARM binary), doesn't necessarily mean that it will work in WebAssembly, as there are various ifdefs which can make certain methods only available in one platform but not another (e.g. the hitWatchdog() invokable from BackendConnection is one such case, I am sure there are also others).

mr-manuel, 2025-03-28:
I think your QML page is locally working, because the versions are different. If you just [compare your code](https://github.com/victronenergy/gui-v2/compare/main...JigneshMali:gui-v2:my-custom-branch) you see that there are a lot of changes from the upstream branch, that you did not apply. It seems that you worked on your QML on the GX device and then tried to copy over the changes to the WASM build, which is by far newer then the GUI version you have on your GX device.

Just make your changes with the latest main branch and also the WASM build will work.

https://github.com/victronenergy/gui-v2/compare/main...JigneshMali:gui-v2:my-custom-branch

mr-manuel, 2025-03-28:
As chriadam said, you find the problems in the developer console of the browser. I tested your wasm build and it's like I said above. Your QML code is not matching the code of the upstream/main branch.

This is only one error you will get, if you don't update your branch with the latest changes:
```
qml: Aborted attempt to push page with errors: /pages/settings/devicelist/battery/PageBattery.qml: qrc:/qt/qml/Victron/VenusOS/pages/settings/devicelist/battery/PageBattery.qml:24 Cannot assign to non-existent property "allowed"
```

[image]

JigneshMali, 2025-03-28:
Thank you ;) [chriadam](https://github.com/chriadam)  & [mr-manuel](https://github.com/mr-manuel). It works now. 
Working on this GUI-V2 for web assembly is a little bit tougher than expected. but now I see what things need to be taken care of when working on this.

closed 2025-03-31 by JigneshMali