victronenergy/gui-v2 #2011: Custom Modifications and Adding a New QML Page for gui-v2 in Venus OS

JigneshMali, 2025-04-01:
Hello everyone,

I’m currently working on some custom modifications to gui-v2 for the Victron Cerbo GX. Specifically, I’m looking into adding a new QML page to enhance the user interface.

Adding a New QML Page:
I want to create a new QML page to display additional data in the UI. How should I go about integrating this page into the existing navigation structure? Any examples or tips on how to handle the interactivity between QML and the backend (such as connecting it to existing services or data)?

DanielMcInnes, 2025-04-01:
Hi @JigneshMali , check out this branch which adds a new qml page: 1947-electric-propulsion-ui-boat-page-v1

JigneshMali, 2025-04-01:
Thank you mr.DanielMcInnes. I missed to add my new qml path to cmakelist.txt.

closed 2025-04-01 by JigneshMali