victronenergy/gui-v2 #2021: Multiphase regulation forced switched to total of all phases when ESS mode is changed

blammit, 2025-04-03:
Multiphase regulation forced switched to total of all phases when ESS mode is changed. See video.

It can be reproduced here:
https://vrm.victronenergy.com/installation/491687/dashboard
https://vrm.victronenergy.com/installation/306244/dashboard
https://vrm.victronenergy.com/installation/496679/dashboard (edited) 

(As per https://victrondevelopment.slack.com/archives/C020QG3JVEW/p1743002722095209)

https://github.com/user-attachments/assets/eb77b0a2-dc0b-4ae2-9bfa-5da25f00355a

mpvader, 2025-04-03:
Izak will look into it

izak, 2025-04-03:
First thing to note: It is not a regression. It also does this in gui-v1 and has done this since at least 2017.

Second: This only happens if you switch to External control and back. Other modes are not affected.

The graphical user interface hides some of the complexity, but what is really happening is that the ESS mode depends on two dbus paths. Specifically, the setting `com.victronenergy.settings/Settings/CGwacs/Hub4Mode` works like this:

1 = Total of phases
2 = Individual phase
3 = External control

And the setting `com.victronenergy.settings/Settings/CGwacs/BatteryLife/State` handles the non-external operation, pretty much all states below 9 are "with batterylife", 9 means keep batteries charged, and above 9 is  "without batterylife". Details are not super important.

So what the old GUI does, and the new one inherited, is it has one control that spans both these settings. And when you set the ESS mode to "external control", it actually sets the hub4mode to 3. Thereby losing the old value. When you reset the ESS mode to something else, it defaults back to 1 (total of phases).

Fixes would be to either split External control out into a separate setting (then it is in hub4control), refactoring the UI so it's not part of the same dropdown, or maybe raising a popup message when the user returns from External control reminding him/her to verify all settings.

I could not find the exact spot in the gui-v2 code where this logic is implemented. But it is almost certainly in the GUI.

mpvader, 2025-04-03:
@blammit, the conclusion/decision from above:

please add a popup/toast with the text, which shows when user changes the ESS setting from External control into something else, which says".

"Make sure to check the Multiphase regulation setting"

we'll do that, only in gui-v2, not gui-v1 and then this issue can be closed.

commit 5f39b188 (2025-05-01): ESS: show toast when changing away from External Control mode

When user changes the ESS mode away from External Control, the
multiphase regulation setting is automatically forced-switched to "total
of all phases". Show a toast notification to inform the user of this
change.

Fixes #2021
closed 2025-05-01 by blammit