victronenergy/gui-v2 #2026: Add support for OrionXS as .dcdc service

ThiemoVanEngelen, 2025-04-03:
[vedirect_interface commit 76b6848](https://github.com/victronenergy/vedirect_interface/commit/76b6848962a9ecc5ac01ac439633e9486d1483ee) adds support for the OrionXS to be connected with its input to the main battery and the output to some auxiliary battery. Whether this is the case or not is determined by a setting in localsettings (`/Setttings/Victron/Products/##serialnr##/OutputBattery`) which is mirrored by the service itself as `/Settings/OutputBattery`. When this setting is 0, there is no change in the behaviour, in the sense that it is presented as a .alternator service. When the setting is 1, the OrionXS will be presented on the dbus as .dcdc service and without the `/Link/*` dbus paths (so DVCC will not interfere). 

Compared to an OrionXS that is configured as PSU (which is also presented as a .dcdc service), an OrionXS configured as charger will have the following extra paths:
`/Settings/OutputBattery` => This will always be 1 for the .dcdc service, but can be set to 0 to configure the OrionXS as being connected to the main battery with its output. When this is changed to 0, it disconnects from the dbus and restarts as a .alternator
`/History/*` => This is contains the charge cycle history, just as when it is a .alternator service

When the OrionXS is configured as .alternator, the `/Settings/OutputBattery` setting is also available and is always 0, but can be changed to 1.

In summary, what needs to be done:

- [ ] Show and allow user to change /Settings/OutputBattery for a .alternator service
- [ ] Conditionally show and allow user to change /Settings/OutputBattery for a .dcdc service
- [ ] Conditionally show the charger history for a .dcdc service

This requires issue #2019 to be solved.

ThiemoVanEngelen, 2025-04-03:
Hi @chriadam 

this is part of a feature which is otherwise (mostly) complete and required to be part of v3.60. welcome to assign it to someone else, this is just to make sure its on your radar.

thanks, Matthijs

ThiemoVanEngelen, 2025-04-03:
having a system for you where you can easily test this on/against is coming.

ThiemoVanEngelen, 2025-04-11:
A system for testing is https://vrm.victronenergy.com/installation/99087/dashboard

ThiemoVanEngelen, 2025-05-06:
The newly introduces setting element should go into a separate "Settings" submenu (where this is then the only item). This settings submenu should not be visible for products where the settings menu would be empty.

The setting itself should be a drop-down with the following options:
- Alternator charging the main battery
- Charging another battery from the main battery

This text is a suggestion but if it does not fit or looks weird, they can be altered. The key is that they are a bit descriptive.

ThiemoVanEngelen, 2025-05-06:
When drop downs are not used anywhere, then it can also be a submenu with a radio button choice.

ThiemoVanEngelen, 2025-05-07:
Examples of the possible layout:

[image]

[image]

[image]

blammit, 2025-05-07:
@ThiemoVanEngelen Looks good - should it be "Setup" instead of "Settings", to align with other pages that use a similar naming? (E.g. for EV chargers, pulse meters, tanks, temperature sensors)

blammit, 2025-05-09:
Actually, some pages use "Settings" and others use "Setup", so I guess it doesn't matter either way.

commit ca1313d3 (2025-04-23): Add support for OrionXS as .dcdc service

The OrionXS can be configured to change from an .alternator service to a
.dcdc service, by changing /Settings/OutputBattery from 0 to 1. When it
is a .dcdc service, it is connected with its input to the main battery
and the output to some auxiliary battery.

To support this:
- allow the /Settings/OutputBattery setting to be changed from
alternator and dcdc settings pages
- since a dcdc service can now have a charger history, move the charger
history implementation from the alternator settings page into a common
DcHistorySettingsColumn that can be shown [... cut]
PR #2058 (2025-04-24): Add support for OrionXS as .dcdc service
The OrionXS can be configured to change from an .alternator service to a .dcdc service, by changing /Settings/OutputBattery from 0 to 1. When it is a .dcdc service, it is connected with its input to the main battery and the output to some auxiliary battery.

To support this:
- allow the /Settings/OutputBattery setting to be changed from alternator and dcdc settings pages
- since a dcdc service can now have a charger history, move the charger history implementation from the alternator settings page into a common DcHistorySettingsColumn that can be shown  [... cut]
commit 715ab507 (2025-04-28): Add support for OrionXS as .dcdc service

The OrionXS can be configured to change from an .alternator service to a
.dcdc service, by changing /Settings/OutputBattery from 0 to 1. When it
is a .dcdc service, it is connected with its input to the main battery
and the output to some auxiliary battery.

To support this:
- allow the /Settings/OutputBattery setting to be changed from
alternator and dcdc settings pages
- since a dcdc service can now have a charger history, move the charger
history implementation from the alternator settings page into a common
DcHistorySettingsColumn that can be shown [... cut]
closed 2025-05-13 by blammit