victronenergy/gui-v2 #2032: Error clearing fails

baudje, 2025-04-07:
running v3.60~57

Clearing errors using the web interface was successful as per below

[image]

However on the local display, an error is shown (that is not shown remotely) and cannot be cleared (nothing happens after pressing the error)

[image]

[image]

chriadam, 2025-04-07:
I will try to reproduce this in the next day or two, next time I'm at the office and have access to a cerbo.  CC @MikeTrahearn-Qinetic @ReinvdZee

MikeTrahearn-Qinetic, 2025-04-19:
I can confirm that when clicking on a Notification delegate, that the correct calls to setting the /Acknowledged flag for the given notification is being set to 1 correctly.

MikeTrahearn-Qinetic, 2025-04-19:
Similar Issue #2042

blammit, 2025-05-01:
Cannot reproduce this, closing.

closed 2025-05-01 by blammit

baudje, 2025-05-01:
do you want me to record and post a video of this? Encountered the issue yesterday again after upgrading to v66. It is again only the BMS connection message that cannot be cleared, clearing other notifications seem to work as intended. Might be linked to using AggregateBattery as BMS. If it is, i'll bring the issue to mr manuel.

ReinvdZee, 2025-05-01:
@baudje,

I've tried to reproduce it here by triggering a lot of alarms and silencing them, but I am unable to reproduce this issue. 

Does this only happen when there were a lot of notifications? Or also with a few?

It the issue persists, I would really like to have a look at your system to see if the backend behaves as expected.