victronenergy/gui-v2 #2035: Translations up- and downloading no longer functional

dirkjanfaber, 2025-04-10:
It seems like the `make download_translations` and `make upload_translations` fell over recently (a couple of weeks ago). 

The script that ups and downloads the translations used to be able to do:

```
git clone https://github.com/victronenergy/gui-v2.git
cd gui-v2
git submodule update --init
mkdir build
cd build
cmake ..
```
Which now fails with:
```
BUILD_DIR_TS_FILES: /data/gui-v2/build/i18n/venus-gui-v2_af.ts;/data/gui-v2/build/i18n/venus-gui-v2_ar.ts;/data/gui-v2/build/i18n/venus-gui-v2_cs.ts;/data/gui-v2/build/i18n/venus-gui-v2_da.ts;/data/gui-v2/build/i18n/venus-gui-v2_de.ts;/data/gui-v2/build/i18n/venus-gui-v2_es.ts;/data/gui-v2/build/i18n/venus-gui-v2_fr.ts;/data/gui-v2/build/i18n/venus-gui-v2_it.ts;/data/gui-v2/build/i18n/venus-gui-v2_nl.ts;/data/gui-v2/build/i18n/venus-gui-v2_pl.ts;/data/gui-v2/build/i18n/venus-gui-v2_ro.ts;/data/gui-v2/build/i18n/venus-gui-v2_ru.ts;/data/gui-v2/build/i18n/venus-gui-v2_sv.ts;/data/gui-v2/build/i18n/venus-gui-v2_th.ts;/data/gui-v2/build/i18n/venus-gui-v2_tr.ts;/data/gui-v2/build/i18n/venus-gui-v2_uk.ts;/data/gui-v2/build/i18n/venus-gui-v2_zh_CN.ts;/data/gui-v2/build/i18n/venus-gui-v2.ts;/data/gui-v2/build/i18n/venus-gui-v2_en.ts
CMake Error at CMakeLists.txt:1252 (add_dependencies):
  Cannot add target-level dependencies to non-existent target
  "venus-gui-v2_lrelease".

  The add_dependencies works for top-level logical targets created by the
  add_executable, add_library, or add_custom_target commands.  If you want to
  add file-level dependencies see the DEPENDS option of the add_custom_target
  and add_custom_command commands.
```

Which seems to be some dependency chain problem.

DanielMcInnes, 2025-04-15:
@dirkjanfaber I think you might be using qt 6.8 by mistake. I can't reproduce this issue with qt 6.6, I can reproduce it with qt 6.8.

dirkjanfaber, 2025-04-15:
Looking at the docker container, that might indeed be the case. I had to rebuild the docker container that runs the script a short while ago. And that one uses `alpine:latest`, which now ships with Qt 6.8.
I need to pin the container to alpine 3.18 and test again.

dirkjanfaber, 2025-04-15:
Needed to use alpine:3.19, but that resolves the issue. Thank you @DanielMcInnes for pointing me into the right direction.
This issue can be closed.

closed 2025-04-15 by dirkjanfaber