victronenergy/gui-v2 #2042: Notifications cannot be acknowledged

ReinvdZee, 2025-04-14:
Seen on the latest Venus OS, (v3.60~58) with the new notification API. In the notification center, there are 3 notifications which cannot be cleared. 

[image]

By the looks of the icon, the notifications seem to be inactive but unacknowledged (unsilenced). Pressing the individual notifications does not do anything and the "Silence All" button is not there. 

I have not been able to reproduce this yet and its unsure whether this is a frontend (gui-v2) or backend (venus-platform) issue.

chriadam, 2025-05-01:
Has anyone been able to reproduce with the most recent release?  Bea changed the implementation of the notifications model, which may have incidentally resolved the issue.  I personally haven't seen this issue at all (even with the previous release).

blammit, 2025-05-01:
Cannot reproduce this, closing.

closed 2025-05-01 by blammit