victronenergy/gui-v2 #2051: RS products: tracker data visibility

martinbosma, 2025-04-18:
The RS and HS products have a setting to enable or disable a PV tracker input. If the tracker is disabled, it should be hidden in the gui.

Related path is `/Pv/n/Enabled`, where n is the tracker index. This also applies to single tracker models.

If `/Pv/n/Enabled` is 0, then the tracker should be hidden and when 1 it should be visible. 
If the item is not present, due to older firmware or unsupported products, then default to visible.

Related Venus issue: https://github.com/victronenergy/venus-private/issues/448

VictronConnect setting:
[image]

Test sites:
- https://vrm.victronenergy.com/installation/55013/dashboard
- https://vrm.victronenergy.com/installation/394634/dashboard

blammit, 2025-05-01:
> If the tracker is disabled, it should be hidden in the gui.

@martinbosma should it be hidden everywhere in the gui? I have implemented it to be hidden in the Overview solar drilldown and the pages that open from that drilldown list, but I'm wondering if these hidden/disabled trackers should still be shown in the Device List settings pages for the solar charger. 

Also, should the Device list settings page also show switches for each tracker to set the /Enabled value, like VictronConnect does in the screenshot?

martinbosma, 2025-05-01:
I'm not sure what you mean with "Device List settings pages for the solar charger" but I'd say, hide it everywhere. And the `/Enabled` item is read-only on the Venus side. Enabling/disabling the trackers in only possible in VictronConnect (for now).

@mpvader Your thoughts?

commit 83ed0757 (2025-05-12): Hide solar trackers with /Enabled=0

A solarcharger, multi or inverter service may set /Pv/<x>/Enabled = 0
(where <x> is the solar tracker identifier) to indicate that the solar
tracker is disabled. If this is the case, hide the tracker in the UI
display.

Ideally QuantityTable would use a filtered model instead of resetting
its colours depending on the row count, but that will be fixed in
future.

Fixes #2051
commit a85a930b (2025-05-16): Hide solar trackers with /Enabled=0

A solarcharger, multi or inverter service may set /Pv/<x>/Enabled = 0
(where <x> is the solar tracker identifier) to indicate that the solar
tracker is disabled. If this is the case, hide the tracker in the UI
display.

Ideally QuantityTable would use a filtered model instead of resetting
its colours depending on the row count, but that will be fixed in
future.

Fixes #2051
closed 2025-05-16 by blammit