victronenergy/gui-v2 #2052: Issue with Custom D-Bus Service in gui-v2.

JigneshMali, 2025-04-21:
i am facing an issue with  custom D-Bus service in gui-v2

The custom D-Bus service com.victronenergy.settings.dummydriver works fine in gui-v1, but isn't trusted or recognized in gui-v2. I get the following error:

### error
`venus-gui-v2.js:8 UID "com.victronenergy.settings.dummydriver/System/NumberofVersions" contains invalid part: "com.victronenergy.settings.dummydriver"`

### code usage part : my qml  page (gui-v2) 
```
import QtQuick
import Victron.VenusOS

Page {
	id: root

     property string bindPrefix: "com.victronenergy.settings.dummydriver"
    property string firmwareUninstalledMessage: qsTr("Driver removed. Please reboot the device.")

	property VeQuickItem FirmwareUninstallationCompleted: VeQuickItem {uid: root.bindPrefix +  "/System/FirmwareUninstallationCompleted"}

	GradientListView {
		model: VisibleItemModel {

			ListText {
				text: "Driver removed. Please reboot the device."  // Directly assigning text as no translation ID exists
				preferredVisible: FirmwareUninstallationCompleted.value === 1  // Control visibility based on your condition
			}

			ListRadioButtonGroup {
				text: "Uninstall  driver"
				dataItem.uid: root.bindPrefix + "/System/FirmwareUninstallconfirm"
				preferredVisible: true
				optionModel: [
					{ display: qsTr("Uninstall"), value: 1 },
					{ display: qsTr("Cancel"), value: 0 }
				]
			}

			ListRebootButton { }
						
		}
	}
}
```

Has anyone encountered this ?

blammit, 2025-04-22:
@JigneshMali add a "dbus/" prefix to the `bindPrefix` so that it is "dbus/com.victronenergy.settings.dummydriver". (The uid passed to VeQuickItem needs the prefix to be a complete uid.)

JigneshMali, 2025-04-22:
@blammit 

```
import QtQuick
import Victron.VenusOS

Page {
	id: root

     property string bindPrefix: "dbus/com.victronenergy.settings.dummydriver"
    property string firmwareUninstalledMessage: qsTr("Driver removed. Please reboot the device.")

	property VeQuickItem FirmwareUninstallationCompleted: VeQuickItem {uid: root.bindPrefix +  "/System/FirmwareUninstallationCompleted"}

	GradientListView {
		model: VisibleItemModel {

			ListText {
				text: "Driver removed. Please reboot the device."  // Directly assigning text as no translation ID exists
				preferredVisible: FirmwareUninstallationCompleted.value === 1  // Control visibility based on your condition
			}

			ListRadioButtonGroup {
				text: "Uninstall  driver"
				dataItem.uid: root.bindPrefix + "/System/FirmwareUninstallconfirm"
				preferredVisible: true
				optionModel: [
					{ display: qsTr("Uninstall"), value: 1 },
					{ display: qsTr("Cancel"), value: 0 }
				]
			}

			ListRebootButton { }
						
		}
	}
}
```

### Issues : 

`UID "dbus/com.victronenergy.settings.dummydriver/System/FirmwareUninstallconfirm" contains invalid part: "dbus"`

blammit, 2025-04-22:
Are you running this in a browser/Wasm? If so, then the uid is mqtt/settings/<x>, with <x> being the device instance of the service.

What do you see when you go to Settings > Debug & Develop > Values? The dummy service should appear there.

JigneshMali, 2025-04-22:
Yes, I am running this in browser. yes in Settings > Debug & Develop > Values, 
the service is active, and it is also working correctly in GUI-V1 as expected. i ma using thie service with gui-v1 without any isseus or conflicts.
only issue is the gui-v2 is not letting this work.

blammit, 2025-04-23:
What is the path to find your service under Settings > Debug & Develop > Values? That will help you to find the correct uid to put in your code. When you load gui-v1 via the browser, this is different as it is not loading it via MQTT/Wasm.

JigneshMali, 2025-04-23:
1. Settings > Debug & Develop > Values > dummysettings > 0 > {then all the list of paths I added (ADD_PATH) in .py service}. 
2. Backend service (.py) is working. No issues with that. 
3. When I use this service in gui-v1, this works and also updates the values in (gui-v2)  Settings > Debug & Develop > Values > dummysettings > 0 > {then all the list of paths} all values get updated. 
4. gui-v2 trust and recognize the service and its all paths and update  them too in  ( Settings > Debug & Develop > Values > dummysettings > 0>)

blammit, 2025-04-23:
> Settings > Debug & Develop > Values > dummysettings > 0 > {then all the list of paths I added (ADD_PATH) in .py service}.

That means when running from the browser (i.e. over Wasm/MQTT) you will need a uid of "mqtt/dummysettings/0" instead. You can call `BackendConnection.type` to check the current backend source type and adjust the uid accordingly.

JigneshMali, 2025-04-24:
1. The error of invalid `"com.victronenergy.settings. dummy driver"` is resolved after adding the "mqtt/ " prefix. It is good that it is accepting the service now.
2. But this ListRadioButtonGroup stopped working as a feature. There are also similar qml pages where my service is running; they are also doing the same.
 ```
ListRadioButtonGroup {
				text: "Uninstall  driver"
				dataItem.uid: "mqtt/com.victronenergy.settings.dummydriver/System/FirmwareUninstallconfirm"
				preferredVisible: true
				optionModel: [
					{ display: qsTr("Uninstall"), value: 1 },
					{ display: qsTr("Cancel"), value: 0 }
				]
			}

```

```
import QtQuick
import Victron.VenusOS

Page {
	id: root

     property string bindPrefix: "mqtt/com.victronenergy.settings.dummydriver"
    property string firmwareUninstalledMessage: qsTr("Driver removed. Please reboot the device.")

	property VeQuickItem FirmwareUninstallationCompleted: VeQuickItem {uid: "mqtt/com.victronenergy.settings.dummydriver/System/FirmwareUninstallationCompleted"}

	GradientListView {
		model: VisibleItemModel {

			ListText {
				text: {
					switch (BackendConnection.type) {
					case BackendConnection.DBusSource:
						return qsTr("Current backend: D-Bus")
					case BackendConnection.MqttSource:
						return qsTr("Current backend: MQTT")
					case BackendConnection.MockSource:
						return qsTr("Current backend: Mock")
					default:
						return qsTr("Backend type: Unknown (%1)").arg(BackendConnection.type)
					}
				}
				preferredVisible: true
			}

			ListText {
				text: "Driver removed. Please reboot the device."  // Directly assigning text as no translation ID exists
				preferredVisible: FirmwareUninstallationCompleted.value === 1  // Control visibility based on your condition
			}

			ListRadioButtonGroup {
				text: "Uninstall  driver"
				dataItem.uid: "mqtt/com.victronenergy.settings.dummydriver/System/FirmwareUninstallconfirm"
				preferredVisible: true
				optionModel: [
					{ display: qsTr("Uninstall"), value: 1 },
					{ display: qsTr("Cancel"), value: 0 }
				]
			}

			ListRebootButton { }
						
		}
	}
}
```

blammit, 2025-04-28:
> But this ListRadioButtonGroup stopped working as a feature. There are also similar qml pages where my service is running; they are also doing the same.

What do you mean that it's stopped working, could you explain?

Also, use BackendConnection.type to use a different uid depending on the backend:

```
VeQuickItem {
    uid: BackendConnection.type === BackendConnection.MqttSource 
        ? "mqtt/dummysettings/0/System/FirmwareUninstallationCompleted"
        : "dbus/com.victronenergy.settings.dummydriver/System/FirmwareUninstallationCompleted"
}
```

JigneshMali, 2025-05-05:
### my backend exmaple code -> 

```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

from dbus.mainloop.glib import DBusGMainLoop
from gi.repository import GLib
import dbus
import dbus.service
import inspect
import pprint
import os
import sys

# our own packages
sys.path.insert(1, os.path.join(os.path.dirname(__file__), '../'))
from vedbus import VeDbusService

softwareVersion = '1.0'

def validate_new_value(path, newvalue):
	# Max RPM setpoint = 1000
	return newvalue <= 1000

def get_text_for_rpm(path, value):
	return('%d rotations per minute' % value)

def main(argv):
		global dbusObjects

		print(__file__ + " starting up")

		# Have a mainloop, so we can send/receive asynchronous calls to and from dbus
		DBusGMainLoop(set_as_default=True)

		# Put ourselves on to the dbus
		dbusservice = VeDbusService('com.victronenergy..settings.dummydriver', register=False)

		# Most simple and short way to add an object with an initial value of 5.
		dbusservice.add_path('/Position', value=5)

		# Most advanced way to add a path
		dbusservice.add_path('/RPM', value=100, description='RPM setpoint', writeable=True,
			onchangecallback=validate_new_value, gettextcallback=get_text_for_rpm)

		# Many types supported
		dbusservice.add_path('/String', 'this is a string')
		dbusservice.add_path('/Int', 0)
		dbusservice.add_path('/NegativeInt', -10)
		dbusservice.add_path('/Float', 1.5)
		dbusservice.add_path('/System/FirmwareUninstallconfirm', 0)

		# Call register after adding paths. More paths can be added later.
		# This claims the service name on dbus.
		dbusservice.register()

		# You can access the paths as if the dbusservice is a dictionary
		print('/Position value is %s' % dbusservice['/Position'])

		# Same for changing it
		dbusservice['/Position'] = 10

		print('/Position value is now %s' % dbusservice['/Position'])

		# To invalidate a value (see com.victronenergy.BusItem specs for definition of invalid), set to None
		dbusservice['/Position'] = None

		print('try changing our RPM by executing the following command from a terminal\n')
		print('dbus-send --print-reply --dest=com.victronenergy.example /RPM com.victronenergy.BusItem.SetValue int32:1200')
		print('Reply will be <> 0 for values > 1000: not accepted. And reply will be 0 for values < 1000: accepted.')
		mainloop = GLib.MainLoop()
		mainloop.run()

main("")
```

###  my QML page code -> 

```
import QtQuick
import Victron.VenusOS

Page {
	id: root

     property string bindPrefix: "mqtt/com.victronenergy.settings.dummydriver"
    property string firmwareUninstalledMessage: qsTr("Driver removed. Please reboot the device.")

	property VeQuickItem FirmwareUninstallationCompleted: VeQuickItem {uid: "mqtt/com.victronenergy.settings.dummydriver/System/FirmwareUninstallationCompleted"}

	GradientListView {
		model: VisibleItemModel {

			ListText {
				text: {
					switch (BackendConnection.type) {
					case BackendConnection.DBusSource:
						return qsTr("Current backend: D-Bus")
					case BackendConnection.MqttSource:
						return qsTr("Current backend: MQTT")
					case BackendConnection.MockSource:
						return qsTr("Current backend: Mock")
					default:
						return qsTr("Backend type: Unknown (%1)").arg(BackendConnection.type)
					}
				}
				preferredVisible: true
			}

			ListText {
				text: "Driver removed. Please reboot the device."  // Directly assigning text as no translation ID exists
				preferredVisible: FirmwareUninstallationCompleted.value === 1  // Control visibility based on your condition
			}

			ListRadioButtonGroup {
				text: "Uninstall  driver"
				dataItem.uid: "mqtt/com.victronenergy.settings.dummydriver/System/FirmwareUninstallconfirm"
				preferredVisible: true
				optionModel: [
					{ display: qsTr("Uninstall"), value: 1 },
					{ display: qsTr("Cancel"), value: 0 }
				]
			}

			ListRebootButton { }
						
		}
	}
}
```

The error of invalid "com.victronenergy.settings. dummy driver" is resolved after adding the "mqtt/ " prefix. It is good that it is accepting the service now.
But this ListRadioButtonGroup stopped working as a feature. There are also similar qml pages where my service is running; they are also doing the same.

[image]

JigneshMali, 2025-05-08:
@blammit 
  can this help?
` import Victron.Mqtt      `

blammit, 2025-05-13:
> But this ListRadioButtonGroup stopped working as a feature. There are also similar qml pages where my service is running; they are also doing the same.

I'm still not sure what you mean by "stopped working as a feature". Can you explain?

> dbusservice = VeDbusService('com.victronenergy..settings.dummydriver', register=False)

I noticed you have two dots in here, should it be 'com.victronenergy.settings.dummydriver' instead?

> can this help?
> import Victron.Mqtt     

If the data is accessible via Debug > Values then it is not necessary to use this import to fetch the data.

JigneshMali, 2025-05-13:
1. The code itself is correct: (it is typo when pasting this here, it is correct in backend.)
`dbusservice = VeDbusService('com.victronenergy.settings.dummydriver', register=False)`

2.I can confirm that the data is visible under Debug > Values. Service is running in the backend. No issues with that.

3.By “stopped working as a feature,” I mean that when I load the dummy driver QML page in GUI-2, the interface no longer responds. For example, a ListRadioButtonGroup on that page won’t let me select any options, and nothing is sent back to the backend.

In contrast, the exact same Python service code works perfectly in GUI-1: I can select radio-button options, the backend updates immediately, and the choices are saved in real time. The problem only occurs with GUI-2.

blammit, 2025-05-13:
> the interface no longer responds

Do you mean, touch events do not work at all? Or is it just that the data is not saved to the backend when you select an option?

JigneshMali, 2025-05-13:
The  touch events work. Like i am  able to open ListRadiobuttonGroup, but the listed options are not being selected as I choose any of them. 
like there are options  : 1. "Uninstall"  2. "Cancel". so when it is default it shows "Unknown" , but in Debug> Value it is at "Cancel" represent by "0".
when I select "Uninstall" it is not being selected. stays at "Unknown"

blammit, 2025-05-13:
Okay, and are you sure the "mqtt/com.victronenergy.settings.dummydriver/System/FirmwareUninstallconfirm" uid for the button group is valid? I am not sure why "com.victronenergy.settings.dummydriver" would work here, because you said that the service settings were found under `Debug > Values > dummysettings > 0` in the UI. If that is the case, I would have expected the uid to be "mqtt/dummysettings/0/System/FirmwareUninstallconfirm".

JigneshMali, 2025-05-15:
**I’ve simplified my service naming for ease of use:**

`Old name: com.victronenergy.settings.dummydriver`
to
`New name: com.victronenergy.dummydriver`

### **With MQTT as our backend, the UID pattern becomes:**
`mqtt/<driver-name>/0/<path>`

**So for example:**
```
Service name: com.victronenergy.dummydriver
Backend connection type: mqtt
Path value: System/FirmwareUninstallconfirm
```

`mqtt/dummydriver/0/System/FirmwareUninstallconfirm`

**Thanks again for the assist, @blammit!**

closed 2025-05-15 by JigneshMali

blammit, 2025-05-15:
Great, glad you were able to work it out :)