victronenergy/gui-v2 #2083: Device List switch page shows inputs as -1

blammit, 2025-04-30:
As per [this comment](https://github.com/victronenergy/gui-v2/pull/2014#issuecomment-2820510028):

When unplugging the device, the outputs in the control card disappear, which is good. When reconnecting the device again with less outputs than before, there are some "-1 | Unsupported type" outputs shown in the device list.

[image]

When reconnecting with less outputs, the paths are not deleted in mqtt explicitly, which may explain this behavior on WASM. However, this also happens on the local GUI.

Reproducible with GIO Extender.

blammit, 2025-05-13:
@ReinvdZee it seems the paths are not deleted on dbus either. The paths do not appear in dbus-spy, but they do appear in `Debug > Values` when running on device. Can that be fixed? If not, I can make a gui-v2 patch to hide any outputs with invalid `/State` values.

chriadam, 2025-05-19:
Discussed in the meeting that we should just hide devices with a state of -1 for now

commit 9ca93447 (2025-05-20): Switch settings: do not show inputs as -1

If an output has an invalid state (e.g. if it is configured as an input)
then do not show it in the list of switch outputs.

Fixes #2083
PR #2151 (2025-05-20): Switch settings: do not show inputs as -1
If an output has an invalid state (e.g. if it is configured as an input) then do not show it in the list of switch outputs.

Fixes #2083
commit 28a1c053 (2025-05-20): Switch settings: do not show inputs as -1

If an output has an invalid state (e.g. if it is configured as an input)
then do not show it in the list of switch outputs.

Fixes #2083
closed 2025-05-20 by blammit