victronenergy/gui-v2 #2089: Enable heatpump role for energy meters

martinbosma, 2025-05-01:
Finishing up the new roles as part of https://github.com/victronenergy/venus-private/issues/380.

The role is already supported, but is read-only, which can now be r/w.

https://github.com/victronenergy/gui-v2/blob/defeebef1e3202c12803fd2d3b3a1fc030d2a19e/pages/settings/devicelist/ac-in/PageAcInSetup.qml#L59-L71

blammit, 2025-05-13:
> We also need a drill down to the heatpump device when the user clicks the load box.

@martinbosma Is that the "AC Loads" or "Essential Loads" box? If it can be either, how do you determine which one?

martinbosma, 2025-05-13:
I'd say a heatpump is an AC load. But perhaps @dirkjanfaber or @izak can comment on that, since they are more involved in the system context.

mpvader, 2025-05-15:
> @martinbosma Is that the "AC Loads" or "Essential Loads" box? If it can be either, how do you determine which one?

@blammit & @martinbosma: that depends on the position. Which is a setting of the meter, exposed in VictronConnect.

blammit, 2025-05-16:
Okay, sounds like this is the same as EVCS, where `/Position=1` means AC-in and "AC loads", and `/Position=0` means AC-out and "Essential Loads".

izak, 2025-05-16:
@martinbosma , remember I asked you about this? And we decided that unless it is asked for we won't change it now?

I added /Position to the cgwacs driver, somewhat by accident, but I don't think it was added to the Victron meter, or the others. So at the moment heat pumps don't have a position.

izak, 2025-05-16:
@blammit, further to my previous comment, heatpump (and acload) services don't have /Position right now, but we're going to add it.

martinbosma, 2025-05-16:
I've added this to vecan-dbus v3.53, which is on the todo but not in a release yet

blammit, 2025-05-20:
I see vecan-dbus v3.53 has now been released as part of Venus v3.60~75, thanks!

blammit, 2025-05-21:
@martinbosma should the "Position" setting be shown (and also editable) for AC inputs with the `heatpump` roles? Currently the setup page only does this for inputs with an `evcharger` role.

izak, 2025-05-21:
@blammit the same logic applies as for the EV charger. If PositionIsAdjustable (or does not exist), then the position can be edited, otherwise just show it.

Showing the total power for each is useful.

Showing the total energy actually isn't all that useful (also not for PV-inverters where it is already done), because it is a total energy counter, and the device may have been installed for years, which makes it a large number that doesn't really tell you what happened today/recently.

We provided for heatpumps to have a temperature, which would be useful to display, except of course  that's only available if you have a real heatpump and not an energy meter proxying for one. So probably not very useful yet.

blammit, 2025-05-22:
Discussed with Izak that the position setting should also be shown for "acload" roles, since /Position is also available for `acload` services now.

closed 2025-05-23 by blammit