victronenergy/gui-v2 #2095: ListCurrentLimitButton currentLimitDialogComponent secondaryText is wrong in some cases

chriadam, 2025-05-05:
From slack: "I have a MultiPlus in my offgrid system (with only AC in 1, and AC in 1 set to Generator) and I noticed that the input current screen incorrectly refers to AC in 2. I have already updated to v3.60~69."

I am not sure what is wrong here, unless the input is 1-based rather than 0-based (our code assumes 0-based input indexes).

The code looks correct to me assuming that the input indexes are zero-based rather than one-based.  i.e. Have secondaryTitle: CommonWords.acInput(root.inputNumber) where the root.inputNumber should be either 0 (for the first generator) or 1 (for the second generator).  The text becomes Ac Input 1  if inputNumber is 0, and becomes AC Input 2 if inputNumber is 1.

chriadam, 2025-05-07:
Raised to high prio in case this is actually causing the current limit to be set for the wrong ac input.
Can be set back to medium prio if it's just a text label thing after all.

blammit, 2025-05-07:
It is just a text label thing, so it won't set the wrong limit, but it's certainly misleading.

blammit, 2025-05-08:
```
// Current limit details on the vebus (or acsystem) service
com.victronenergy.vebus.<suffix>/Ac/ActiveInput = 0
com.victronenergy.vebus.<suffix>/Ac/In/1/CurrentLimit
com.victronenergy.vebus.<suffix>/Ac/In/1/CurrentLimitIsAdjustable

// Service name, source, etc. for the input, as reported on the system service
com.victronenergy.system/Ac/In/0/ServiceName
com.victronenergy.system/Ac/In/0/Source

// Service type, as configured by user settings (indicating Grid/Shore/Generator type)
com.victronenergy.settings/Settings/SystemSetup/AcInput1
```

The codebase needs to be more explicit about whether a 0-based or 1-based numbering is being used when translating the "AC input <x>" text.

commit d654f758 (2025-05-08): Show correct AC input number in when changing current limit

The digit used to represent an AC input may be 0-based or 1-based
depending on the context. For example, the first AC input current limit
is configured by com.victronenergy.vebus.<suffix>/Ac/In/1/CurrentLimit
while the same input's service name is available as
com.victronenergy.system/Ac/In/0/ServiceName.

The current limit dialog showed a 0-based AC input title string instead
of the correct 1-based number. Fix this and replace
CommonWords.acInput() with acInputFromIndex() and acInputFromNumber() to
make it more explicit as to whethe [... cut]
commit 21879b8a (2025-05-08): Show correct AC input number in when changing current limit

The digit used to represent an AC input may be 0-based or 1-based
depending on the context. For example, the first AC input current limit
is configured by com.victronenergy.vebus.<suffix>/Ac/In/1/CurrentLimit
while the same input's service name is available as
com.victronenergy.system/Ac/In/0/ServiceName.

The current limit dialog showed a 0-based AC input title string instead
of the correct 1-based number. Fix this and replace
CommonWords.acInput() with acInputFromIndex() and acInputFromNumber() to
make it more explicit as to whethe [... cut]
PR #2111 (2025-05-08): Show correct AC input number in when changing current limit
The digit used to represent an AC input may be 0-based or 1-based depending on the context. For example, the first AC input current limit is configured by com.victronenergy.vebus.<suffix>/Ac/In/1/CurrentLimit while the same input's service name is available as com.victronenergy.system/Ac/In/0/ServiceName.

The current limit dialog showed a 0-based AC input title string instead of the correct 1-based number. Fix this and replace CommonWords.acInput() with acInputFromIndex() and acInputFromNumber() to make it more explicit as to whether [... cut]
commit 26dc9b3c (2025-05-08): Show correct AC input number in when changing current limit

The digit used to represent an AC input may be 0-based or 1-based
depending on the context. For example, the first AC input current limit
is configured by com.victronenergy.vebus.<suffix>/Ac/In/1/CurrentLimit
while the same input's service name is available as
com.victronenergy.system/Ac/In/0/ServiceName.

The current limit dialog showed a 0-based AC input title string instead
of the correct 1-based number. Fix this and replace
CommonWords.acInput() with acInputFromIndex() and acInputFromNumber() to
make it more explicit as to whethe [... cut]
closed 2025-05-08 by blammit