victronenergy/gui-v2 #2131: Switching pane: backend synchronization issues

ReinvdZee, 2025-05-13:
Recently, the sliders in the switching pane were modified to write the value to the backend and set the slider state equal to the value read from the backend, so all instances of the GUI always show the correct value. This seemed to work fine, but now issues do arise.

When dragging the slider back and forth a couple of times, it starts to lag behind, which is not very nice. If then a digital output is also switched on/off a few times, it enters an oscillating state it cannot exit without restarting the backend:

https://github.com/user-attachments/assets/ecf007d2-586e-4a0f-a8cd-ef6027ff11fe

This reproduces on the GX IO extender and on the Node red virtual switch.
State and dimming value persistance was added (not merged yet) to the GX IO Extender, which writes the new state/dimming value to localsettings on every change, increasing latency.

mman, 2025-05-13:
Good catch, this needs a `debounce` trigger to only send the last changed value say after 250ms when it became stable.

ReinvdZee, 2025-05-13:
I improved the dbus-switch backend by only writing the dimming value to localsettings whenever it has been stable for a second.

closed 2025-05-20 by blammit