victronenergy/gui-v2 #2133: Improve "Display DC Loads" text

philipptrenz, 2025-05-13:
The subtext "Calculated estimate of DC consumption" caused some confusion, because it is actually not correct in case a SmartShunt is used to measure the DC consumption.

Therefore, the subtext "Calculated estimate of DC consumption" should just be removed, so that onle the "Display DC Loads" label remain on this entry.

[image]

philipptrenz, 2025-05-14:
Hi @chriadam, I remember that we also discussed a wording like "Measured or estimated DC loads" during the settings redesign project and finally decided against it, as it might raise questions that are not being answered within the gui. Let's do as suggested above: Removing the subtitle and only keep the primary label.

chriadam, 2025-05-15:
Merged to main, closing as fixed.

closed 2025-05-15 by chriadam