victronenergy/gui-v2 #2135: Switching pane: Multiple cards of the same group

ReinvdZee, 2025-05-14:
Found on Venus OS 3.60~73. When restarting a `com.victronenergy.switch` service which has multiple outputs in a single group, the group is displayed multiple times: 

[image]

This only reproduces with virtual switches and the GX IO Extender, not with the ES Smart switch.

This requires the switch service to store the group value, which was added recently for the GX IO extender (dbus-switch) and node-red. Both are not merged in Venus OS yet. So this is blocked until that is merged or we (@dirkjanfaber and I) prepare something for qinetic to easily reproduce it.

dirkjanfaber, 2025-05-21:
At the moment I am unable to re-produce this on 3.60~76.

ReinvdZee, 2025-05-21:
I also tested this on 3.60~76 and can still reproduce this, but the behavior changed a bit. 

At first, I set two outputs to have the same group. After reconnecting the GX IO Extender, two groups showed up again. Then I changed the group name of both outputs from "Victron" to "Group 1". This seems to be successful, as it is shown correctly on dbus and on the setup page of the device page. However, the new group does not show up. Instead, I see a control card with the old group name "Victron", with just one output, but two switch icons in the top-left of the card:

[image]

Outputs 1 & 2 are configured to be part of the group "Group1": 

[image]

And output 1 is not present in any of the control cards.

ReinvdZee, 2025-05-21:
This issue can also be reproduced with the ES Smart switch on Venus OS 3.60~76:

[image]

And afterwards changing a group name also is not reflected in the switching pane.

blammit, 2025-05-21:
I'm not able to reproduce this with both the GIO Extender and ES Smart Switch on 3.60~75 and 3.60~76, so perhaps this is triggered by some timing/syncing issue around when the groups are added and removed.

ReinvdZee, 2025-05-22:
Somehow, I also no longer can reproduce this. Closing the issue for now, we can reopen if it pops up again.

closed 2025-05-22 by ReinvdZee