victronenergy/gui-v2 #2160: MultiPlus is shown as Unknown in device menu, when off

mr-manuel, 2025-05-22:
It should display `Off` instead.

[image]

[image]

Like on the overview page:

[image]

DanielMcInnes, 2025-11-20:
See 'Cerbo HQ2243 (Echten)'

chriadam, 2025-11-20:
This bug might have been fixed by https://github.com/victronenergy/gui-v2/pull/2587 (see the resolved task: https://github.com/victronenergy/gui-v2/issues/2564)

blammit, 2025-11-20:
@izak do you know what might be happening here? We could put a workaround in the UI to check the /Mode as well and show "Off" in the case that the /State is not set, but ideally it wouldn't be needed, and also I don't know if this affects other services with a /State and /Mode  as well, like `inverter` and `acsystem`.

mpvader, 2025-11-27:
@vejpasop , if I remember correctly, this is because when the mode is set to off, the unit doesn't respond and then technically, mk2dbus isn't sure wether the device is off or not because its not telling Venus OS that its off.

But, when state is off, how about just changing Unknown to off on the driver (= mk2dbus) level?

And @martinbosma how does this work for RS/vecan-dbus?

martinbosma, 2025-11-27:
For the RS the comms remains active if the mode is set to off (e.g. via GX or VictronConnect).
However, there are a few cases where the unit drops comms, i.e. will disappear from dbus:
- When the main power switch is set to off
- Remote on/off switch (on aux connector) set to off and no solar/grid
- Wired BMS (on aux connector) signals battery is empty and no gird/solar

chriadam, 2026-03-02:
@mr-manuel with the fix mentioned above in mk2-dbus, is this still presenting as a bug in gui-v2?  If so, can someone tell me precisely what we need to do to fix this one, as I'm a bit lost?