victronenergy/gui-v2 #2197: Cellular icon is not visible on Wasm

blammit, 2025-05-30:
As per https://victrondevelopment.slack.com/archives/C06EVSJDR3K/p1748107654508389:

"Today I noticed that the cellular icon is only visible on the GX screen, but not on the WASM. The WiFi icon, on the other hand, is displayed on both."

The icon should be shown if `/SimStatus` is vavlid for the modem service.

blammit, 2025-06-01:
Reported for https://vrm.victronenergy.com/installation/366384/ but the `SimStatus` is not valid there when viewed via Wasm. Needs further investigation.

philipptrenz, 2025-06-03:
Hi @blammit, thanks for looking into it. On dbus, it's `1000` and I see the same value on MQTT.

[image]

mpvader, 2025-06-05:
as discussed, Bea will check again.

blammit, 2025-06-13:
Actually, the instance=1 was from a recording, and discussed with Izak that the `modem` instance number should always be zero, as the `modem` service has no actual DeviceInstance. Philipp has confirmed this bug does not occur with a real hardware device. Closing.