victronenergy/gui-v2 #2200: Add third option for current/watts preference

mpvader, 2025-05-31:
[image]

But first lets see what other feedback comes after release.

mpvader, 2025-11-06:
https://community.victronenergy.com/t/v3-70-beta-amps-vs-watts-questions-and-discussion/39183/8

philipptrenz, 2025-11-13:
To summarize the community thread:

- It's about the "Electrical power display" setting, implemented by PR #138
- Seemingly, we have initially introduced the Amp vs. Watt setting with a bug, which caused it to only affect the AC side – this was perceived by users as a feature
- Some people are exclusively asking for the battery to show Amps instead of Watts (maybe because only having a battery as DC component)
- Others asking for having Watts for the AC side and Amps for DC side, which is what we originally had before the configurable option

Easiest solution, as suggested in the title, is adding a third option next to "Power (Watts)" and "Current (Amps)", which allows to show the AC-side in Power and the DC-side in Current and therefore restore the behavior prior to. It would therefore also be an option to make this new setting the default.

As a title I would propose: `Mixed (AC in Watts, DC in Amps)`

[image]

ReinvdZee, 2025-11-13:
Note, this is stored in this setting: `/Settings/Gui/ElectricalPowerIndicator`, directly written to by the GUI. To allow a third option (with value 2), and to set the default value, [this line](https://github.com/victronenergy/meta-victronenergy/blob/master/meta-ve-software/recipes-ve/gui-v2-webassembly/localsettings#L7) needs to be updated:

philipptrenz, 2025-11-26:
@blammit localsettings is updated and changes are available from v3.70~61: https://github.com/victronenergy/meta-victronenergy/commit/0b45e5ee40a2bbd46f0fa73e92e8cf81680386bc

chriadam, 2025-12-12:
Bea is on vacation for a while, so I'll take over polishing and finishing this one, even though I have far less understanding of all this stuff than Bea does.

For the above issue: can you provide a fake screenshot of what you'd expect to see on the DC Loads page when there is no dcsystem service, according to your suggestion?  Is it just: the top table would not exist, so only the bottom two delegates would exist, and both of those would display Current instead of Total Power?  Or am I misunderstanding?

And does this also apply to the AC Loads for the single-phase case?

chriadam, 2025-12-15:
I have updated the PR in order to implement this, I believe.  Could you provide a couple of installations I can test with, to make sure I've done it properly?  (I also must be allowed to change the setting between "prefer amps", "prefer watts", and "mixed", so hopefully not a customer installation.)

closed 2025-12-19 by chriadam