victronenergy/gui-v2 #2201: Look into watchdog timeout tripping on local lan

mpvader, 2025-06-02:
When trying to open gui-v2 WASM on local LAN, having the browser set to 3G throttling, you can never open it.

A watchdog timeout expires causing a reload; so there is never enough time to fully download the wasm file.

Perhaps the browser/javascript can see that a download is ongoing and not stuck - and then not trip?

Cause to look into this: https://community.victronenergy.com/t/new-ui-issues-due-to-low-bandwidth/34224/7

I'll first discuss this with Dirk-Jan

chriadam, 2025-06-02:
Indeed.  We should only start the watchdog timeout timer after gui-v2 has fully loaded the first time.  I assume it would be simple for gui-v2 to set some "initialized" js variable, which the watchdog timeout timer would check for prior to triggering the reload() path.

FWIW, Bart lists one other possible option: "option to pre-download the UI blob from a fast source (like it’s done by VRM)." - this one should definitely be possible too, as we already support passing e.g. cerbo MQTT broker address via query parameter to the wasm blob etc.  Some significant extra complexity exists though: e.g. use a HTTP GET of the wasm.sha256 first, ask VRM if it has that one and if so download it from there, otherwise download "normally" from the device, etc.

DanielMcInnes, 2025-06-05:
[image]

commit b1039e2b (2025-06-13): Disable the watchdog from reloading wasm builds until initialized.

This gives enough time for slow connections to load wasm builds.
Fixes #2201
commit a45993df (2025-06-14): Disable the watchdog from reloading wasm builds until initialized.

This gives enough time for slow connections to load wasm builds.
Fixes #2201