victronenergy/gui-v2 #2203: Custom GUI v2 Not Reflected in VRM Remote Console (via VRM Logger (https://vrm.victronenergy.com/installation/XXXXX/dashboard))

JigneshMali, 2025-06-03:
Hi everyone,

I’m currently running a Cerbo GX with a customized GUI v2. I’ve made some UI changes and use it both via HDMI display and through WASM-based Remote Console over SSH—everything works perfectly in those environments.

However, when I access the same Cerbo GX system via the Remote Console on the VRM Portal (through the VRM logger ((https://vrm.victronenergy.com/installation/XXXXX/dashboard))), I notice that the GUI there does not reflect my custom GUI v2 changes. Instead, it shows the default standard GUI-v2.

I’m wondering:

- Why doesn’t the VRM Remote Console reflect my customized GUI v2?
- Is there any caching or version mismatch happening on the VRM logger side?
- How can I ensure that my custom GUI is visible when accessing via VRM?

Thanks in advance for any guidance!

mr-manuel, 2025-06-03:
Check this code

https://github.com/victronenergy/gui-v2/blob/4872dc958293371a7362091a62aa3e8bda357507/scripts/build-wasm.sh#L131-L134

Don't know how you build your own modded version, but you need to also upload the `venus-gui-v2.wasm.sha256` of your wasm build. After that you need to restart the vrmlogger service, so it notifies VRM that your wasm has a different checksum that on VRM exists. This will trigger VRM to download the wasm directly from your GX device when opening. Additionally you have to reload the VRM page, since it makes this check only on the first page load and not on changing pages.

```bash
svc -t /service/vrmlogger
```

Else just reboot your GX device.

I saw also your issue https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/268#issuecomment-2933973227 here. This brings a bit confusion to me. Are you using the modifications that are installed by dbus-serialbattery or your own?

JigneshMali, 2025-06-03:
Hi @mr-manuel ,

 I have two Venus OS setups:
1st : Raspberry Pi (Venus OS) – Here ,I’m using the latest official version of the dbus-serialbattery driver as-is, without any modifications.

2nd : Cerbo GX – This setup has custom modifications to the dbus-serialbattery driver tailored to my use case.

On both devices, I’ve noticed that the VRM UI doesn’t match what’s shown on the local HDMI display or in the Remote Console.

mr-manuel, 2025-06-03:
1st: Raspberry Pi does not support GUIv2 on an attached screen yet

2nd: There are different build processes for the attached screen and the WASM build, [check the Wiki](https://github.com/victronenergy/gui-v2/wiki/How-to-build-venus-gui-v2#building-using-the-build-scripts).

JigneshMali, 2025-06-04:
> 1st: Raspberry Pi does not support GUIv2 on an attached screen yet

Yes , Raspberry pi does not support attached screen. yesterday  i tried to reomve installation from VRM and again adding it to VRM. Also rebooting the Device. Now VRM console and GUI-v2 wasm on browser are same. It may requierd a clean add installation on VRM again and device reboot. it is working now.

> 2nd: There are different build processes for the attached screen and the WASM build, [check the Wiki](https://github.com/victronenergy/gui-v2/wiki/How-to-build-venus-gui-v2#building-using-the-build-scripts).

For this i hv to investigate where it is not working,

### my device path  (/var/www/venus/gui-v2) has these files -> 
```
icons (directory)
index.html
LICENSE.txt
Makefile
qtloader.js
venus-gui-v2.js
venus-gui-v2.wasm.gz
venus-gui-v2.wasm.sha256
victronenergy.svg
```

dirkjanfaber, 2025-06-04:
Sorry for breaking in and being a bit pedantic about this. The issue tracker within gui-v2 is for actual issues within in gui-v2, not for the integration within VRM and modifications of the gui. Those discussions are best kept in the modification area within our community. 
You have already started a thread there, so lets continue it there: https://community.victronenergy.com/t/custom-gui-v2-not-reflected-in-vrm-remote-console-via-vrm-logger/34392.

And lets not keep the developers of the gui-v2 occupied by issues they shouldn't be involved with. So closing this issue.