victronenergy/gui-v2 #2208: On a managed battery, number of blocking modules shows invalid

izak, 2025-06-06:
[image]

What I know so far, it seems to get lost somewhere between the "details" object, and the page displaying it, because hardcoding a value in the details page like this:

```
        VeQuickItem {
            id: nrOfModulesBlockingCharge
            //uid: root.bindPrefix + "/System/NrOfModulesBlockingCharge"
            value: 22
        },
```

Doesn't do anything. So issue is in one of the other components. Still looking.

izak, 2025-06-06:
Bingo! The bug is that `QuantityObject` shows `--` if no `unit` is defined.

blammit, 2025-06-09:
Thanks for tracking this down. The Units::getDisplayTextWithHysteresis() code is incorrect and should be removed - it was written with the assumption that if there is a valid number value, then the unit will not be `Units_None`. 

We should also make the QuantityObject `unit` property a required property, instead of defaulting to `Units_None`, so that it is always explicitly set. That can be fixed separately in the `main` branch.

commit 3f52d967 (2025-06-09): Show quantity number even when unit is Units_None

It is valid for a number value to not have a unit (i.e. unit is
Units_None). In this case, show the number without a unit suffix,
instead of showing "--".

This bug was causing "--" to be shown on the battery details page under
"Number of modules blocking charge / discharge", instead of showing the
number value.

Also, set the decimals for the QuantityObject so that the number is
not shown with unnecessary decimals. Do this for the cell id values in
PageBatteryDetails.qml as well.

Fixes #2208
commit 7a4fc336 (2025-06-09): Show quantity number even when unit is Units_None

It is valid for a number value to not have a unit (i.e. unit is
Units_None). In this case, show the number without a unit suffix,
instead of showing "--".

This bug was causing "--" to be shown on the battery details page under
"Number of modules blocking charge / discharge", instead of showing the
number value.

Also, set the decimals for the QuantityObject so that the number is
not shown with unnecessary decimals. Do this for the cell id values in
PageBatteryDetails.qml as well.

Fixes #2208
closed 2025-06-09 by chriadam