victronenergy/gui-v2 #2222: UI freezes until restarted

mpvader, 2025-06-11:
[image]

https://www.facebook.com/groups/victronenergyusergroup/permalink/1097697575505181/

lets discuss in dev meeting how to approach this

mpvader, 2025-06-13:
same person posted here:

https://community.victronenergy.com/t/v3-60-cerbo-screen-freezes-un-responsive/35677/4

mpvader, 2025-07-23:
Per v3.70~16 we've updated to Qt 6.8.3, nice.

Thanks for all involved - as usual this is not as easy as we all like it to be.

So now back to this issue: I've asked the users in that thread on Community to install v3.70~19 and let us know if all works well for them now.

mpvader, 2025-07-26:
Someone fixed a certain type of freeze by replacing his GX and/or GX Touch

https://community.victronenergy.com/t/randomly-frozen-gx-touch-display-on-cerbo-gx/32782/10

chriadam, 2025-11-13:
Closing as we cannot reproduce.  If we find a repro, we will reopen for investigation.

closed 2025-11-13 by chriadam