victronenergy/gui-v2 #2237: WASM: Add option to set language via url / parameters

mpvader, 2025-06-13:
Question from a support department:

> Is it somehow possible to open the new ui remote console in the language the user has set? > instead of the language the system is set to?

chriadam, 2025-06-16:
I had this PR https://github.com/victronenergy/gui-v2/pull/458 to implement this (i.e. accept a language as a query parameter, which could be set by e.g. VRM, and then that query parameter would override the system setting).

However, ISTR we discussed in a meeting that it might confuse the user (e.g. they might think that the language being shown WAS the system setting language).  If that is not a concern, I can update that above PR, and then the next step would be to find out how we can get that language setting from the VRM environment.

mpvader, 2025-06-19:
discussed today: this will prbly indeed confuse users.

chriadam, 2025-11-06:
Pushing this back to v1.3.0 for now.  We will probably end up closing this task as WontFix but let's see.

chriadam, 2026-09-17:
Let's at least prototype this properly, for discussion, as it's useful for installers to be able to help fix customer issues, even if the customer uses a language unknown to the installer.

1) the language selection from argument / query param must be used INSTEAD OF any system setting for language
2) the language settings screen should clearly show what the system setting language is, but that it's overridden in the current app instance due to argument / query param
3) the system setting for language should be read-only when display language has been overridden via argument / query param
4) changes to the system for language should be ignored by app instances which are displaying a specific language specified via argument / query param (i.e. so that another user accessing the device and toggling between Danish and German doesn't cause an installer's app instance to switch away from English, for example, if the installer used a query param to force display via English).
5) ... perhaps some other considerations (e.g. need some VRM-side plumbing to allow installers to launch remote console with a specific language param?)